egzumer/uv-k5-firmware-custom #2: Battery percent

OneOfEleven, 2023-10-02:
Hi egzumer,

Someone's pointed out that the full charge lithium battery charge voltage is 8.4V (which is correct). But your little percent function is saying 8.15V is 100% full. I've done the battery cal using a DMM so the voltage is correct (it was actually quite a bit out), that's assuming the battery on the back doesn't have any series diodes to the two gold terminals at the bottom of the battery (when I measure the battery voltage from).

Braccato, 2023-10-03:
Ciao 1of11 sul tuo firmware, l'icone della batteria rimane ferma ? le tacche non scendono?

egzumer, 2023-10-03:
This is on purpose not to rise question "why my radio doesn't charge to 100%?", and to have it showing nice 100% after USB charging finishes. The USB charger charges to 8.1-8.2 so it will never charge to the real 100%, but charging it to 8.4 from 8.15 adds very little to the charge state. The error it causes happens only at the highest state of charge and is minimal.

The battery terminals at the back show the real battery voltage, these are shorted to those on the radio side.

egzumer, 2023-10-03:
> Ciao 1of11 sul tuo firmware, l'icone della batteria rimane ferma ? le tacche non scendono?

The battery notches are calculated differently, they use original implementation and their levels are written in the EEPROM. That is why they are different than percentage.

egzumer, 2023-10-03:
I'm open to changing this to the real state of charge of 8.4. I can do the modification. Just note the above. Also to add to that, barely any charger will charge the battery to 8.4V really. It usually charges until current drops to e.g. 50mA and switches of, immediately after that the voltage at the battery terminals drops to well below 8.4

OneOfEleven, 2023-10-03:
No it's OK, you're totally right about the last couple 100 mV, leave as is - well for me anyway. I spose if the muggles gang together and send a written protest in, then we can still ignore em ;)

closed 2023-10-03 by egzumer

lohtse, 2023-10-03:
> No it's OK, you're totally right about the last couple 100 mV, leave as is - well for me anyway. I spose if the muggles gang together and send a written protest in, then we can still ignore em ;)

LOL yes just ignore them please....