egzumer/uv-k5-firmware-custom #4: Fagci spectrum analyzer sometimes hangs

pawol, 2023-10-03:
Fagci spectrum analyzer sometimes hangs and makes QS not responding until power reboot

egzumer, 2023-10-03:
I did play with it quite a bit, didn't notice any problems. Can you describe it in more details. Do you have to change any settings to trigger the hang or does it hang after a while after turning it on without any changes?

pawol, 2023-10-03:
withing total time  of 1h of playing with analyzer I got the issue at least 3x.
I did nothing special, just sumple exploration if keys' functions, scrolling, etc.

egzumer, 2023-10-03:
I'm afraid I won't be able to fix that then. First of all I just put these firmware's together, I don't have that much insight into what's going on there on the lower level, and once per 15 or more min is to infrequent to track it down. Maybe there is some pattern of actions or some particular settings that trigger this, if you track it down it would make things much easier. You may also want to try the unmodified FW from fagci's repository for comparison. I use the the [refactor](https://github.com/fagci/uv-k5-firmware-fagci-mod/tree/refactor) branch.

lohtse, 2023-10-03:
Not sure if you have seen but theres heen a few fixes to spectrum and is now on repo

pawol, 2023-10-03:
I understand your POV. To be honest I am not interested too in Fagci solution - rather not necessary in real live :)

egzumer, 2023-10-03:
> Not sure if you have seen but theres heen a few fixes to spectrum and is now on repo

I did see there are new commits, although I don't know which branch to use. I used the `refactor` one as that one was the last and most frequently updated. I'll ask which one to use.

If anyone want to test that here are the original fagci FW files:
`fagci_main_dc3aeab8.packed.bin` - main branch so probably the one intended for release
`fagci_refactor_ffe1573a_iused.packed.bin` - the one that I used
`fagci_refactor_fae171f3_newest.packed.bin` - updates applied
[compiled-firmware.zip](https://github.com/egzumer/uv-k5-firmware-custom/files/12796988/compiled-firmware.zip)

lohtse, 2023-10-03:
> I did see there are new commits, although I don't know which branch to use. I used the `refactor` one as that one was the last and most frequently updated. I'll ask which one to use.
>
> If anyone want [...]

Was more about giving you a heads up incase the fixes solved the issues mention here.

I personally only ever build using refactor branch when compiling

As to the supplied zip I will wait till you add fixes as prefer to compile  with my own requirements but thank you for providing the zip...

egzumer, 2023-12-03:
I have suspicion that it has something to do with an external antenna. Some induced, relatively high voltage may be the cause. I don't use one so I cannot confirm nor deny, but I had some reports that the radio comes back to life when antenna is unscrewed, or that radio starts to play voices on a firmware that has that feature disabled (voices use the same pins as buttons).

kamilsss655, 2023-12-22:
Not sure if related by I've noticed very rare freezes when exiting from spectrum. Wondering how that would be possible..

DarioGCQ, 2024-01-30:
Hi egzumer,
i love your FW version but the Spectrum part is old.... The fagci FW have a Spectrum better, fast and without stop problem!
I try to merge the two but the new spectrum.c have to many difference and i can't compile it without error!!

You have in program to upgrade your spectrum in the future?
Thank you for your work!
Ciao

egzumer, 2024-02-01:
> You have in program to upgrade your spectrum in the future?

No

bsdinis1, 2024-02-17:
If the selected VFO is in 'Frequency Mode', specifically in the 'F7+' band, exiting the 'Spectrum Analyzer' causes the selected VFO to stop showing the current band (F7+).
I tried to reproduce this issue multiple times, and there was one instance where the radio froze and I had to reboot it.

bsdinis1, 2024-02-17:
The green LED above the display remains on when exiting the 'Spectrum Analyzer' with an open squelch.