egzumer/uv-k5-firmware-custom #7: The possibility of switching the scan list using f+7

CaptainRi, 2023-10-04:
VOX is not commonly used, can we use f+7 for toggles scanlist-1 on/off for current channel when held down

lohtse, 2023-10-04:
> VOX is not commonly used, can we use f+7 for toggles scanlist-1 on/off for current channel when held down

or may be toggle between scanlist one and two?

CaptainRi, 2023-10-04:
> > VOX is not commonly used, can we use f+7 for toggles scanlist-1 on/off for current channel when held down
> 
> or may be toggle between scanlist one and two?

yes

Yag00000, 2023-10-04:
f+1 or f+2 are free !!!!

egzumer, 2023-10-05:
Implemented by @OneOfEleven. Long press `Scan` button while scanning. Already [released](https://github.com/egzumer/uv-k5-firmware-custom/releases/tag/v0.7).

closed 2023-10-05 by egzumer