egzumer/uv-k5-firmware-custom #8: minor issue ;-)

lohtse, 2023-10-04:
-custom-2/ -I D:/TWatch/uv-k5-firmware-custom-2//external/CMSIS_5/CMSIS/Core/Include/ -I D:/TWatch/uv-k5-firmware-custom-2//external/CMSIS_5/Device/ARM/ARMCM0/Include -c app/main.c -o app/main.o
In file included from app/main.c:28:
D:/TWatch/uv-k5-firmware-custom-2/app/spectrum.h:127:12: error: padding struct to align 'frequencyChangeStep' [-Werror=padded]
  127 |   uint32_t frequencyChangeStep;
      |            ^~~~~~~~~~~~~~~~~~~
D:/TWatch/uv-k5-firmware-custom-2/app/spectrum.h:151:1: error: padding struct size to alignment boundary [-Werror=padded]
  151 | } ScanInfo;
      | ^
D:/TWatch/uv-k5-firmware-custom-2/app/spectrum.h:159:1: error: padding struct size to alignment boundary [-Werror=padded]
  159 | } RegisterSpec;
      | ^
D:/TWatch/uv-k5-firmware-custom-2/app/spectrum.h:165:12: error: padding struct to align 'f' [-Werror=padded]
  165 |   uint32_t f;
      |            ^
cc1.exe: all warnings being treated as errors
make: *** [Makefile:326: app/main.o] Error 1

might be a problem lol

lohtse, 2023-10-04:
new error buddy

FIX -DENABLE_FASTER_CHANNEL_SCAN -DENABLE_RSSI_BAR -DENABLE_AUDIO_BAR -DENABLE_COPY_CHAN_TO_VFO  -I D:/TWatch/uv-k5-firmware-custom-2/ -I D:/TWatch/uv-k5-firmware-custom-2//external/CMSIS_5/CMSIS/Core/Include/ -I D:/TWatch/uv-k5-firmware-custom-2//external/CMSIS_5/Device/ARM/ARMCM0/Include -c ui/main.c -o ui/main.o
ui/main.c: In function 'UI_DisplayAudioBar':
ui/main.c:94:4: error: this 'if' clause does not guard... [-Werror=misleading-indentation]
   94 |    if (gCurrentFunction != FUNCTION_TRANSMIT ||
      |    ^~
ui/main.c:104:5: note: ...this statement, but the latter is misleadingly indented as if it were guarded by the 'if'
  104 |     const unsigned int voice_amp  = BK4819_GetVoiceAmplitudeOut();  // 15:0
      |     ^~~~~
cc1.exe: all warnings being treated as errors
make: *** [ui/main.o] Error 1
Press any key to continue . . .

egzumer, 2023-10-05:
The one before was glitch in the matrix :). 
The second is overly constraining compiler settings. I don't get those errors. You can use compile-with-docker script if you have docker installed.

lohtse, 2023-10-05:
so what you are saying is even though it has vompiled perfectly fine before I now have to install FURTHER software to compile Even though I have ALWAYS used the Win_make.bat

lohtse, 2023-10-05:
and don't forget these are YOUR instructions

[image]

lohtse, 2023-10-05:
and also the last issue was the same as oneofeleven whish has now been resovled by them... you willeven see  posted a comment/issue regarding it

lohtse, 2023-10-05:
on investigaion

if this is disable the issue occurs

ENABLE_TX1750                 := 1

OneOfEleven, 2023-10-05:
Github still makes my brain hurt lol.

I'm see'ing/getting messages intended for egzumers version but addressed to me, having trouble deciding who the intended recipient is.

Right, gona be an easy going day for me, yesterday was crazy busy.

lohtse, 2023-10-05:
> Github still makes my brain hurt lol.
> 
> I'm see'ing/getting messages intended for egzumers version but addressed to me, having trouble deciding who the intended recipient is.
> 
> Right, gona be an easy going day for me, yesterday was crazy busy.

lol this is egzummer  

you fixed the issue in your FW lol

And yes  have an easy day or better still take a day off!!!!!!!!!!!!!

closed 2023-10-09 by egzumer