egzumer/uv-k5-firmware-custom #11: SQL

Electronicpilot, 2023-10-05:
The SQL threshold is too high even by 1.

subsubl, 2023-10-05:
Sensitiviti seems ok for me, local 2M repeater in fact is constantly opening on SQL 1 had to switch to 2.

Electronicpilot, 2023-10-05:
The sensitivity is okay, but the response threshold is 12dB different from the stock one, and when working in the direct channel or when moving, this is not comfortable. What is still well understood by ear is cut off by a noise suppressor.

Electronicpilot, 2023-10-05:
This is not just my opinion.

egzumer, 2023-10-05:
People do different things, some FWs do it in code. I think the correct way is adjusting eeprom calibration entries. There is a python tool for that.
https://github.com/amnemonic/Quansheng_UV-K5_Firmware/tree/main/python-utils
`squelch_table_read.py`
`squelch_table_write.py`

and a good manual (google translate is your friend):
[Guida Firmware Squelch V_batt.pdf](https://github.com/egzumer/uv-k5-firmware-custom/files/12822994/Guida.Firmware.Squelch.V_batt.pdf)

subsubl, 2023-10-05:
The squelch table that Matoz spm81 uses improved one where SATCOM and weaker airband opens normally. And SQL has 9 steps, no point of using it on 1. Could be even higher for low noise areas?

ik5wwp, 2023-10-10:
> People do different things, some FWs do it in code. I think the correct way is adjusting eeprom calibration entries. There is a python tool for that. [...]

I have some problems to translate "Fine del pippone" ... :-D
I'm the autor of the manual! :-D :-D

egzumer, 2023-10-10:
> I have some problems to translate "Fine del pippone" ... :-D

Me too :)

> I'm the autor of the manual! :-D :-D

Good job sir :)

By the way, the firmware now includes 1o11's enhencements in SQL sensitivity in the code. I switched to sql 4 and it is still more sensitive than old 1.

ik5wwp, 2023-10-10:
> By the way, the firmware now includes 1o11's enhencements in SQL sensitivity in the code. I switched to sql 4 and it is still more sensitive than old 1.

Yes, I noticed that: with this new firmwares there is no need to adjust eprom entries.
By the way, whats happened to 1of11 repository?

egzumer, 2023-10-10:
I read that he had some oopsy with his recent updates, he pushed some code that erases calibration data from radio. Should be back soon.

closed 2023-10-11 by egzumer