egzumer/uv-k5-firmware-custom #13: Spectrum

rai4424, 2023-10-05:
Spectrum is not properly working onUhf band.

lohtse, 2023-10-05:
> Spectrum is not properly working onUhf band.

what frq?? as working for me!

lohtse, 2023-10-05:
> Spectrum is not properly working onUhf band.

tried throughtout 400 - 470MHz. and all seems fine too me buddy

lohtse, 2023-10-05:
Some what unrelated to OPs issue but related to spectrum...

Latest build when you change band type using 0/fm key  the fw locksup /crashes only way to fix is reboot radio..

rai4424, 2023-10-05:
On 462Mhz after pressing forward back I noticed that it was not receiving signal of my other radio from 100 feet distence.

egzumer, 2023-10-05:
I just copied the spectrum code I don't have any insight in what is going on there, you can try to load the authors FWs [available here](https://github.com/egzumer/uv-k5-firmware-custom/issues/4#issuecomment-1745745735). If the problem is still there you can try to report that to him.

The other problem is that the newest spectrum code has grown in size, maybe they already fixed it, but I'm no longer able to implement it here because it won't fit anymore.

closed 2023-10-08 by rai4424