egzumer/uv-k5-firmware-custom #9: I'm very sorry for not expressing myself clearly.

CaptainRi, 2023-10-05:
I'm very sorry for not expressing myself clearly.

When you were integrating the OneOfEleven firmware, the addition of the scan list function in the  OneOfEleven firmware “**keypad-5**/NOAA button now toggles scanlist-1 on/off for current channel when held down - IF NOAA not used” **was covered by fagci spectrum analyzer (F+5 to turn on)**, so I was wondering if you could use F+keypad-7 to complete toggles scanlist-1 on/off for current channel when held down, as the VOX function is not commonly used.

In addition, keypad-8 or 9 are not very commonly used and can be used to arrange other functions

Also, I think your firmware is currently the best, thank you very much!

OneOfEleven, 2023-10-05:
I sometimes use the 8/R button to have a quick listen on the repeater input frequency. Whether to use the 5/Noaa or 7/vox key for channel scanlist toggle and/or spectrum totally depends on if the user has NOAA or vox enable, it's an awkward thing to decide really.

lohtse, 2023-10-05:
> I sometimes use the 8/R button to have a quick listen on the repeater input frequency.
> 
> Whether to use the 5/Noaa or 7/vox key for channel scanlist toggle and/or spectrum totally depends on if the user has NOAA or vox enable, it's an awkward thing to decide what to do really.

the current use of the * button which is normally used anyway for scanning is actually perfect, also notice the pt1 will change scanlist too which is nice too.

OneOfEleven, 2023-10-05:
I'll move the channel scanlist toggle to 7/vox key then (take me so time to get used to lol).

lohtse, 2023-10-05:
Bugt what happens then if some one uses VOX? like many do?

OneOfEleven, 2023-10-05:
That's the problem.

What to do for the best, aarrgghh lol

OneOfEleven, 2023-10-05:
What's pt1 ?

lohtse, 2023-10-05:
> What's pt1 ?

button one below ptt

OneOfEleven, 2023-10-05:
ah

Mine toggles the squelch on/off (monitor)

lohtse, 2023-10-05:
> ah
> 
> Mine toggles the squelch on/off (monitor)

mine is set to start scan pt2 is long press Monitor short is fm

OneOfEleven, 2023-10-05:
fagci uses the flashlight side button to start the bandscope/spectrum display, but it got changed to 5/noaa key.

lohtse, 2023-10-05:
> fagci uses the flashlight side button to start the bandscope/spectrum display, but it got changed to 5/noaa key.

may be you could use that?

OneOfEleven, 2023-10-05:
right, at the moment, I've set the channels scanlist toggle to use ..

* key-7 if VOX is disabled, otherwise ..
* key-5 if NOAA is disabled, otherwise ..
* back to having to use the menu way to change the setting

lohtse, 2023-10-05:
to be honest I am little unsure why the OP does not just set pt1 to scanlist same as me(CPS and chirp does this by defualt!) and then just use the new toggle function to move between scanlists

lohtse, 2023-10-05:
you start scan by pressing p1  and you press exit to stop... that simple!!

OneOfEleven, 2023-10-05:
I got lower side button set to start/stop scan

oh my head again lol

lohtse, 2023-10-05:
> I got lower side button set to start/stop scan
> 
> oh my head again lol

so thats actualy what the OP is wanting!

egzumer, 2023-10-05:
I started using different functions for long press and for F+number. In my implementation the F+5 is spectrum analyzer and long press 5 is scan toggle.

lohtse, 2023-10-05:
default scan options using side buttons always scans on the channel you are current using and thats what the OP is wanting to do... I see NO reason for the OP to not use the side button for this function as by default it is already there..

lohtse, 2023-10-05:
> I started using different functions for long press and for F+number. In my implementation the F+5 is spectrum analyzer and long press 5 is scan toggle.

eeerm  long press is not starting scan on  latest build(as of today)

OneOfEleven, 2023-10-05:
Have I broke your favourite button ?

lohtse, 2023-10-05:
> Have I broke your favourite button ?

whos???    if so nope   not yet. and actaully you made my fave buttons better with the scan functions you added. even pressing pt1(sidebutton) actually changes scanlists so thats awesome too...

egzumer, 2023-10-05:
> eeerm long press is not starting scan on latest build(as of today) well for me anyway

Longpress 5 is toggle scan lists for current channel, not start scan. * is start scanning and toggle lists.

I read it too quickly and didn't understand the initial post it correctly. I like the current arrangement we should start using different function for longpress and F+ more

lohtse, 2023-10-05:
> > eeerm long press is not starting scan on latest build(as of today) well for me anyway
> 
> Longpress 5 is toggle scan lists for current channel, not start scan. * is start scanning and toggle lists.
> 
> I read it too quickly and didn't understand the initial post correctly. I like the current arrangement we should start using different function for longpress and F+ more
yeah  eeerm "Longpress 5 is toggle scan lists for current channel"  that does not work no matter who long I press it... but thats fine as pressing * will change scanlists

OneOfEleven, 2023-10-05:
Sounds good.

How to remember what buttons/sequences do what will be the main problem I think - put radio down for a week or so then come back to it ;)

lohtse, 2023-10-05:
> Sounds good.
> 
> How to remember what buttons/sequences do what will be the main problem I think - put radio down for a week or so then come back to it ;)

will soon need  new readme file with all the new button combos lol

egzumer, 2023-10-05:
> yeah eeerm "Longpress 5 is toggle scan lists for current channel" that does not work no matter who long I press it... but thats fine as pressing * will change scanlists

https://github.com/egzumer/uv-k5-firmware-custom/assets/14902414/fc7961e4-7d3b-4d24-9378-e2d40f0a4c27

lohtse, 2023-10-05:
Arrr shit you mean it adds current channel to scanlist

OneOfEleven, 2023-10-05:
I just moved that to 7/vox key lol, though it could move back to 5/noaa key - all depends on user setting

egzumer, 2023-10-05:
> How to remember what buttons/sequences do what will be the main problem I think 

Remembering all the shortcuts is still better than not having them at all. I'm fine with that until someone implements voice recognition :)

lohtse, 2023-10-05:
> I just moved that to 7/vox key lol, though it could move back to 5/noaa key - it's dependent on user setting

well for me I use VOX with a throat mic so it would be an issue

egzumer, 2023-10-05:
Just to clarify because I had trouble understanding what the OP means and he did not respond yet. I guess he understood that toggling scanlist assignment was covered with spectrum, it is not. It is longpress and F+. So I guess the original request is unjustified.
Using longpress 5 for other function solves me one other issue, because holding 5 for too long causes the spectrum analyzer to enter frequency setting function and fills it with "555555555".

OneOfEleven, 2023-10-05:
I'll fix any/all problems like that once I add the spectrum in, I keep going off on a tangent, mainly bug hunting.

CaptainRi, 2023-10-05:
> 然后，我将频道扫描列表切换移动到 7/vox 键（花点时间让我习惯哈哈）。

非常抱歉困扰到您。因为我是菜鸟不会更改按键功能，如果会改的话就不会给您造成困扰了。
另外是在 app/main.c 里修改相应值就能完成功能的修改吗？

closed 2023-10-09 by CaptainRi