egzumer/uv-k5-firmware-custom #17: Display gone during Tx

rai4424, 2023-10-07:
A serious isue has been noticed. Battery voltages drops immediately during Tx and resumes after. My Talkie's display vanished two times.  After that I installed stock firmware, there was no such issue.

rai4424, 2023-10-08:
A number of videos recorded by me.

rai4424, 2023-10-08:
I  have re-checked privious FWs, v.07, and v.06 were showing battery voltage drop issue while TX. So I m on v.04 with sad face.

Yag00000, 2023-10-08:
battery drop is correct in TX in real time

rai4424, 2023-10-08:
battery drop is correct in TX in real time , due to high consumption in TX

No its special case. Voltage drop in huge and display 
was glitching. Disply was totally white two times on 80% Battery. I have  recording but cant post hare.

rai4424, 2023-10-08:
Screen shot 1

rai4424, 2023-10-08:
Still not resolved the issue

lohtse, 2023-10-08:
I use this formware and have been ysing custom daily builds for two weeks and NEVER once had the claimed issue....

rai4424, 2023-10-08:
I use this formware and have been ysing custom daily builds for two weeks and NEVER once had the claimed issue....

This custom FW was released two days ago, how you are using it for two weeks?

lohtse, 2023-10-08:
I made the effort to build/compile when new commits have been made..hence as stated custom build..I enable and add more then in statdard build so as a result would experince issues if tuere was any.

lohtse, 2023-10-09:
I wonder it the issue you are experiencing is hardware related or due a conflict in the prebuilt FW  you are using.  Like I have already stated I have this Firmware running on my K5s(6 in total)  and not one has had issue as per here..  But I am using one I have compiled myself with all the latest's fixes etc.. SO that could actually be the reason why. Below is what I have  enabled 
[image]

This is what is enabled in prebuilt
[image]

egzumer, 2023-10-09:
This commit [149a0f5](https://github.com/egzumer/uv-k5-firmware-custom/commit/149a0f5bdc962c93d073d9756e14f64ed1520e30) blocks voltage reading when radio is transmitting (like it used to be)

rai4424, 2023-10-09:
> This commit [149a0f5](https://github.com/egzumer/uv-k5-firmware-custom/commit/149a0f5bdc962c93d073d9756e14f64ed1520e30) blocks voltage reading when radio is transmitting (like it used to be)

You are working great، I m waiting for v0.9 eagerly

rai4424, 2023-10-09:
May be this issue woll be resolved in next version

rai4424, 2023-10-09:
Thanks you very much

closed 2023-10-09 by rai4424