egzumer/uv-k5-firmware-custom #23: NOT an issue just a curiousity!

lohtse, 2023-10-10:
I understand from some of your comments it may not be possible to add any new Spectrum versions.... So this is just a curiosity question more than anything. SO please do not see this as and issue, or request.

Is therer a possiblily to add the options seen in the screen shot
[image]

They have been in the now removed refactor spectrum version for some time and curious as to the possibility of having them in Your FW  as they would be useful.

Again d not take this as a request just curious if possible..

thank you for building the FW,

egzumer, 2023-10-10:
You mean DEV, CMP, MIC and MIX?  I don't know how complicated it would be, but sure it is possible. As I said before I didn't look too closely at what is going on inside, it was relatively easy to integrate it with just a copy and paste, plus changing a few lines of code. But with a recent changes it is getting too complicated to synchronize projects. Especially after changes of 1o11's in the naming of variables and so on.

Also 1o11 is working on his own adaptation of spectrum analyzer so it doesn't make sense for me to do the same thing, I'm nowhere near his skills, I'm can only help out a little bit.

lohtse, 2023-10-10:
> You mean DEV, CMP, MIC and MIX? I don't know how complicated it would be, but sure it is possible. As I said before I didn't look too closely at what is going on inside, it was relatively easy to integrate it with just a copy and paste, plus changing a few lines of code. But with a recent changes it is getting too complicated to synchronize projects. Especially after changes of 1o11's in the naming of variables and so on.
> 
> Also 1o11 is working on his own adaptation of spectrum analyzer so it doesn't make sense for me to do the same thing, I'm nowhere near his skills, I can only help out a little bit.

NO worries and thank you for replying.   was just curious if "You mean DEV, CMP, MIC and MIX? " could be done...

closed 2023-10-10 by egzumer