egzumer/uv-k5-firmware-custom #24: v0.9 radio cannot unlock

CaptainRi, 2023-10-10:
After entering the automatic lock keyboard of the radio, pressing and holding the F key cannot unlock it.

lohtse, 2023-10-10:
> After entering the automatic lock keyboard of the radio, pressing and holding the F key cannot unlock it.

may be something wrong with your radio... saw you comment and checked. double check and triple checked both precompiled and my own compiled versions and works as it should

CaptainRi, 2023-10-10:
Besides the version I compiled myself, using the egzumer compiled version also has the same problem @egzumer

lohtse, 2023-10-10:
> Besides the version I compiled myself, using the egzumer compiled version also has the same problem @egzumer

nope body on telegram has reported the issue   either.. Also as this is 1of11s base FW  no issue with that too...    I honestly believe your radio is faulty.... An unlocking issue would be widely reported in the telegram group as 1of11, egzumers are two out of the three MAIN fws used and NO one has reported the issue.... I have 2 radios with custom versions of egzumers FW and 4 with 1of11s flashed and none have issues unlock OTHER than when using scan mde wher it happens but this has been a  bug since the stock firmwares

lohtse, 2023-10-10:
https://github.com/egzumer/uv-k5-firmware-custom/assets/48331155/019a1fe6-7cfa-4b32-b9ac-c1cf8db010f4

no issues and that radio has been on and in use for two hours

CaptainRi, 2023-10-10:
Your statement is very arbitrary. I have two K5 machines, and others have also reported this issue to me, which already exists in previous versions. The faulty machine just flashed back to the official firmware and did not encounter this issue, so it's not my radio issue

lohtse, 2023-10-10:
> Your statement is very arbitrary. I have two K5 machines, and others have also reported this issue to me, which already exists in previous versions. The faulty machine just flashed back to the official firmware and did not encounter this issue, so it's not my radio issue

1300 in a group over well over half using egzumer and 1of11 and no one reported the issue.. Also video above showing no issue... mmmmmmmmmmmmm  but yeeeeeeeeeeeeeeah theres a bug that only effects a few devices I guess

egzumer, 2023-10-10:
@CaptainRi I don't see this issue either. You can try to download dump configuration and calibration with [k5prog-win](https://github.com/OneOfEleven/k5prog-win/raw/main/k5prog_win.exe) and attach this here together with the FW bin that you use. Maybe it is configuration dependent.

CaptainRi, 2023-10-10:
[https://github.com/CaptainRi/video/blob/main/er.mp4?raw=true](url)

lohtse, 2023-10-10:
> @CaptainRi I don't see this issue either. You can try to dump configuration and calibration with [k5prog-win](https://github.com/OneOfEleven/k5prog-win/raw/main/k5prog_win.exe) and attach it here together with the FW bin that you use. Maybe it is configuration dependent.

he has now reported the issue on telegram.. HE FORGOT to mention it was when he USED FMRADIO..That I can confirm has an issue..

lohtse, 2023-10-10:
[image]

egzumer, 2023-10-10:
One can turn off the FM radio and then unlock the keyboard not really an issue. Also the other way around you have to lock the keyboard and then use side button function to turn the FM radio on. I don't see any other way.

lohtse, 2023-10-10:
> One can turn off the FM radio and then unlock the keyboard, not really an issue. Also the other way around you have to lock the keyboard and then use side button function to turn the FM radio on to be unable to unlock it. I don't see any other way.

talking him through setting a side button for fmradio on telegram.

egzumer, 2023-10-10:
But how does he turn on the FM radio when locked, without a side button?

lohtse, 2023-10-10:
> But how does he turn on the FM radio when locked, without a side button? Or how to lock the radio if FM is acthe turns it on before it locks

egzumer, 2023-10-10:
But I can't lock it when the FM is on.

egzumer, 2023-10-10:
Ok it locks on autolock, then it is an issue, I'll look at it.

CaptainRi, 2023-10-10:
By the way, versions released before October 4th do not have this issue，Can you handle this problem？
Anyway, thank you very much！

lohtse, 2023-10-10:
> Ok it locks on autolock, then it is an issue, I'll look at it.

I have reported the issue to 1of11 too

egzumer, 2023-10-10:
Fixed, binary if you want to test:
[firmware.packed.zip](https://github.com/egzumer/uv-k5-firmware-custom/files/12857210/firmware.packed.zip)

closed 2023-10-10 by egzumer

lohtse, 2023-10-10:
seen youn have updated GH  testing now.

and is working

commit 64bc29d3 (2023-10-10): FIX #24: lock/unlock doesn't work when FM radio active