egzumer/uv-k5-firmware-custom #26: TX

Romashca, 2023-10-11:


egzumer, 2023-10-11:
I don't have such a problem. Does it work like that when the plug is disconnected? Where did you get the firmware?

Romashca, 2023-10-11:
it works the same way with the plug disconnected, the firmware from your repo

egzumer, 2023-10-11:
But did you build your own or did you use the release provided, which version?

Romashca, 2023-10-11:
collected in vision studio code , today from repo

lohtse, 2023-10-11:
am wondering which Frqs,  I di have similar issue if AM band is one of the two selected Frqs on screen. And if AM is being received or just finished it can be a second or tow before I can trans on my other frq..

Romashca, 2023-10-11:
yes it is am

lohtse, 2023-10-11:
> yes it is am

then thats the actually issue... I am pretty sure because the Radios chips do funky things to get AM it takes time to switch back for FM... I notice the same issue on Stock FW's  and all other custom ones too but is a little worse on customs possible due to the AM fixes employed...

Romashca, 2023-10-11:
with other messages the same trouble

lohtse, 2023-10-11:
Ple

> yes, it 's Am , it's the same at closed frequencies, the inscription does not go out

Please don't forget that devs are making these radios do things they are not designed for.

> with other messages the same trouble

what other messages?

egzumer, 2023-10-11:
All you do is press the PTT and the message stays on permanently? Every time?

Romashca, 2023-10-11:
with other messages , the same trouble , for example , busy on a busy channel ,yes every time

egzumer, 2023-10-11:
Did you try with the release provided?
https://github.com/egzumer/uv-k5-firmware-custom/releases

Romashca, 2023-10-11:
not yet

egzumer, 2023-10-11:
When dual watch is in use, the last channel that was receiving becomes an active channel for tx for couple of second, no issue there.

The issue supposedly is that the massage stays permanently.

lohtse, 2023-10-11:
> When dual watch is in use, the last channel that was receiving becomes an active channel for tx for couple of second, no issue there.
> 
> The issue supposedly is that the massage stays permanently.

he needs to provide videos  to be fair...

lohtse, 2023-10-11:
> When dual watch is in use, the last channel that was receiving becomes an active channel for tx for couple of seconds, no issue there.
> 
> The issue supposedly is that the massage stays permanently.

I never really saw it as an issue.. I accepted the radios limitations new and old ;-)

Romashca, 2023-10-11:
https://github.com/egzumer/uv-k5-firmware-custom/assets/83151488/4f6bf0be-d339-40c4-a7b5-44378ed13962

lohtse, 2023-10-11:
> Untitled.video.-.Made.with.Clipchamp.1.mp4

so thats the same as what I posted above

Romashca, 2023-10-11:
it is worth paying attention to the fact that the second channel is already receiving and this one is still blocked

Romashca, 2023-10-11:
you press transfer after the inscription has disappeared, but I have reception on the second channel

lohtse, 2023-10-11:
watch it properly... it is rx when I first press   then I do it after the RX has finished also

Romashca, 2023-10-11:
is it really so difficult to see and understand this

Romashca, 2023-10-11:
watch the video carefully and find the differences

lohtse, 2023-10-11:
we will have to agree to disagrre as I see the exact same situation..

and egzumer has stated this which is correct

When dual watch is in use, the last channel that was receiving becomes an active channel for tx for couple of seconds, no issue there.

egzumer, 2023-10-11:
The problem stated was that the message stays permanently, not that it appears on the other channel.

Romashca, 2023-10-11:
from your words, does it mean that there is no problem that the message hangs until the radio restarts ?  because it can hang for a long time?

Romashca, 2023-10-11:
finally at least someone understood

lohtse, 2023-10-11:
> from your words, does it mean that there is no problem that the message hangs until the radio restarts ? because it can hang for a long time?

custom build or prebuild fw?

lohtse, 2023-10-11:
> finally at least someone understood

I was trying to fully understand and even replicate you issue... I could NOT replicate it....  Thats isnot , not understanding it is trying to reproduce a fault someone has to confirm YOUR issue buddy.. I could not do that.

lohtse, 2023-10-11:
> > finally at least someone understood
> 
> I was trying too fully understand and even replicate your issue... I could NOT replicate it.... Thats is not , not understanding it is trying to reproduce a fault someone has to confirm YOUR issue buddy.. I could not do that.

Romashca, 2023-10-11:
everything is written above

Romashca, 2023-10-11:
as a tester, I described the problem, provided a video, I don't care at all if you are indifferent to the problem

lohtse, 2023-10-11:
> everything is written above

what frqs are bing used as one is hidden and the video does not show all the info..

You state everything is written above.  yet yesterday someone had an issue unlocking their radio.. theyb failed to explain that it only happened during FM radio use. as they just stated when using the radio.. I could not reproduce their issue UNTILL all the facts became apparent.

lohtse, 2023-10-11:
> as a tester, I described the problem, provided a video, I don't care at all if you are indifferent to the problem

I am literally trying to reproduce your DAMN issue so that it can be confirmed and help the dev.

Romashca, 2023-10-11:
10 mb of video doesn 't give the full picture ? make it so that I can provide the video as a whole

lohtse, 2023-10-11:
> 10 mb of video doesn 't give the full picture ? make it so that I can provide the video as a whole

try recording it fully and using this site to compress it as thats what I do

https://www.freeconvert.com/video-compressor/download

lohtse, 2023-10-11:
and provide the FRQ ranges thats always hidden in the vids.   as well as confirm  fw version or if you compiled and setup used incase there is an actual setting conflict. even zip the fw.bin file and upload so the dev can take a look

Romashca, 2023-10-11:
[firmware.zip](https://github.com/egzumer/uv-k5-firmware-custom/files/12868706/firmware.zip)

lohtse, 2023-10-11:
> [firmware.zip](https://github.com/egzumer/uv-k5-firmware-custom/files/12868706/firmware.zip)

prebuild? custom built? which version? setup info??   I have been using and compiling egzumer for around two weeks and not had the issue... So the more info you provide the better it is for egzumer to help... in the mean time go to firmware channel and try the custom builds there and see if the issue continues and provide a vide etc

Romashca, 2023-10-11:
air frequencies, low band, you don't know which frequencies are closed in the repo?

lohtse, 2023-10-11:
> air frequencies, low band, you don 't know which frequencies are closed in the repo , how about busy lock ? what about high voltage? these messages hang until you restart the fucking radio

I am done with you... Am trying to help but you are not getting what you want do being abusive..  Godd luck and hope your radio dies

Romashca, 2023-10-11:
downloaded the repository https://github.com/egzumer/uv-k5-firmware-custom ,compiled in visual studio code

Romashca, 2023-10-11:
долбоёб конченный

Romashca, 2023-10-11:
я надеюсь что ты сдохнешь от рака мозга!

egzumer, 2023-10-11:
Try the relese i provided

lohtse, 2023-10-11:
I assume this я надеюсь что ты сдохнешь от рака мозга! пидор! will be deleted given that this is what he said I hope you die of brain cancer! fag!

Romashca, 2023-10-11:
relese is good. all work.

egzumer, 2023-10-11:
Then provide what you change

Romashca, 2023-10-11:
the release works, but I'm not used to asking and would like to change some data myself, font, etc.

Romashca, 2023-10-11:
> Затем укажите, что вы изменяете

where should I specify this ?

Romashca, 2023-10-11:
> Я предполагаю это, я надеюсь, что ты сдохнешь от рака мозга! ! будет удалено, учитывая, что это то, что он сказал, я надеюсь, что вы умрете от рака мозга! пидор!
it was addressed to you
https://github.com/lohtse

egzumer, 2023-10-11:
If the only thing you change is makefile then provide the build option that you choose. You should have started with that. Think again, how am I suppose to track it down without it?

 If you made source code changes then you caused the bug that you have hunt down on your own.

lohtse, 2023-10-11:
> If the only thing you change is makefile then provide the build option that you choose. You should have started with that. Think again, how am I suppose to track it down without it?
> 
> If you made source code changes then you caused the bug that you have hunt down on your own.

Literally what I was trying to get too... hence asking for build , if user compiled, setup used etc.

Romashca, 2023-10-11:
I change the makefile from your repo , I don 't touch the source code , the output is what I wrote about

Romashca, 2023-10-11:
> > Если единственное, что вы изменяете, это makefile, укажите выбранный вами вариант сборки. С этого надо было начать. Подумайте еще раз, как я могу отыскать его без него?
> > Если вы внесли изменения в исходный код, значит, вы вызвали ошибку, которую вам пришлось искать самостоятельно.
> 
> Буквально то, что я тоже пытался получить... Следовательно, запрашивается сборка, если пользователь скомпилирован, настройка используется и т. Д.

Go love yourself, I'm communicating with a person

lohtse, 2023-10-11:
> I change the makefile from your repo , I don 't touch the source code , the output is what I wrote about

but if you provide the setup used it can help identify if it was a setting  that was enable or even disable that caused your issue!  Thats why I asked you for that info.

lohtse, 2023-10-11:
> Go love yourself, I'm communicating with a person

There is no need for that at all...

lohtse, 2023-10-11:
> #define ONE_OF_ELEVEN_VER
> 
> #ifdef GIT_HASH #define VER GIT_HASH #else #define VER "231004" #endif
> 
> #ifndef ONE_OF_ELEVEN_VER const char Version[] = "OEFW-"VER; const char UART_Version[] = "UV-K5 Firmware, Open Edition, OEFW-"VER"\r\n"; #else const char Version[] = "1o11+fagci-"VER; const char UART_Version[] = "UV-K5 Firmware, Open Edition, 1o11+fagci-"VER"\r\n"; #endif

the info fro the make file  not version file, version file will not actually give  version it defines how to write it on boot screen!

Romashca, 2023-10-11:
# compile options (see README.md for descriptions)
# 0 = disable
# 1 = enable
#
ENABLE_CLANG                  := 0
ENABLE_SWD                    := 0
ENABLE_OVERLAY                := 0
ENABLE_LTO                    := 1
ENABLE_UART                   := 1
ENABLE_AIRCOPY                := 1
ENABLE_FMRADIO                := 0
ENABLE_NOAA                   := 0
ENABLE_VOICE                  := 0
ENABLE_VOX                    := 0
ENABLE_ALARM                  := 0
ENABLE_TX1750                 := 1
ENABLE_PWRON_PASSWORD         := 0
ENABLE_BIG_FREQ               := 0
ENABLE_SMALL_BOLD             := 0
ENABLE_KEEP_MEM_NAME          := 1
ENABLE_WIDE_RX                := 1
ENABLE_TX_WHEN_AM             := 0
ENABLE_F_CAL_MENU             := 1
ENABLE_CTCSS_TAIL_PHASE_SHIFT := 1
ENABLE_BOOT_BEEPS             := 0
ENABLE_SHOW_CHARGE_LEVEL      := 1
ENABLE_REVERSE_BAT_SYMBOL     := 1
ENABLE_CODE_SCAN_TIMEOUT      := 0
ENABLE_AM_FIX                 := 1
ENABLE_AM_FIX_SHOW_DATA       := 0
ENABLE_SQUELCH_MORE_SENSITIVE := 1
ENABLE_FASTER_CHANNEL_SCAN    := 1
ENABLE_RSSI_BAR               := 1
ENABLE_AUDIO_BAR              := 0
ENABLE_COPY_CHAN_TO_VFO       := 1
ENABLE_SPECTRUM               := 1
#ENABLE_SINGLE_VFO_CHAN        := 1
#ENABLE_BAND_SCOPE             := 1

#############################################################

Romashca, 2023-10-11:
[Makefile.zip](https://github.com/egzumer/uv-k5-firmware-custom/files/12869321/Makefile.zip)
[firmware.zip](https://github.com/egzumer/uv-k5-firmware-custom/files/12869322/firmware.zip)

[17 more replies]

commit 7e6a78ab (2023-10-11): FIX #26: vfo state (TX DISABLE label) update issue when FM radio disabled in Makefile
closed 2023-10-11 by egzumer