egzumer/uv-k5-firmware-custom #32: A proposal for a new font.

PieeerB, 2023-10-17:
There is a ready-made font from mbg, maybe it can be used in the firmware. I attach screenshots and the font itself. Of course, everything is at your discretion.
[image]
[image]
[image]
[image]
[image]
[font.c.txt](https://github.com/egzumer/uv-k5-firmware-custom/files/12964826/font.c.txt)

PieeerB, 2023-10-17:
`/* Copyright 2023 Dual Tachyon
 * https://github.com/DualTachyon
 *
 * Licensed under the Apache License, Version 2.0 (the "License");
 * you may not use this file except in compliance with the License.
 * You may obtain a copy of the License at
 *
 *     http://www.apache.org/licenses/LICENSE-2.0
 *
 *     Unless required by applicable law or agreed to in writing, software
 *     distributed under the License is distributed on an "AS IS" BASIS,
 *     WITHOUT WARRANTIES OR CONDITIONS OF ANY KIND, either express or implied.
 *     See the License for the specific language governing permissions and
 *     limitations under the License.
 */

#include "font.h"

//const uint8_t gFontBig[95][16] =
const uint8_t gFontBig[95][15] =
{
	{0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00},
	{0x00, 0x00, 0x00, 0xF8, 0xF8, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x1B, 0x1B, 0x00, 0x00},
	{0x00, 0x1E, 0x3E, 0x00, 0x00, 0x3E, 0x1E, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00},
	{0x40, 0xF0, 0xF0, 0x40, 0xF0, 0xF0, 0x40, 0x00, 0x04, 0x1F, 0x1F, 0x04, 0x1F, 0x1F, 0x04},
	{0xE0, 0xF0, 0x10, 0x1C, 0x1C, 0x30, 0x20, 0x00, 0x08, 0x19, 0x11, 0x71, 0x71, 0x1F, 0x0E},
	{0x60, 0x60, 0x00, 0x00, 0x80, 0xC0, 0x60, 0x00, 0x18, 0x0C, 0x06, 0x03, 0x01, 0x18, 0x18},
	{0x00, 0xB0, 0xF8, 0xC8, 0x78, 0xB0, 0x80, 0x00, 0x0F, 0x1F, 0x10, 0x11, 0x0F, 0x1F, 0x10},
	{0x00, 0x20, 0x3E, 0x1E, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00},
	{0x00, 0x00, 0xE0, 0xF0, 0x18, 0x08, 0x00, 0x00, 0x00, 0x00, 0x07, 0x0F, 0x18, 0x10, 0x00},
	{0x00, 0x00, 0x08, 0x18, 0xF0, 0xE0, 0x00, 0x00, 0x00, 0x00, 0x10, 0x18, 0x0F, 0x07, 0x00},
	{0x00, 0x00, 0x40, 0xC0, 0x80, 0xC0, 0x40, 0x00, 0x00, 0x01, 0x05, 0x07, 0x03, 0x07, 0x05},
	{0x00, 0x00, 0x00, 0xC0, 0xC0, 0x00, 0x00, 0x00, 0x00, 0x01, 0x01, 0x07, 0x07, 0x01, 0x01},
	{0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x20, 0x3C, 0x1C, 0x00, 0x00},
	{0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x01, 0x01, 0x01, 0x01, 0x01, 0x01, 0x01},
	{0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x18, 0x18, 0x00, 0x00},
	{0x00, 0x00, 0x00, 0x00, 0x80, 0xC0, 0x60, 0x00, 0x18, 0x0C, 0x06, 0x03, 0x01, 0x00, 0x00},
	{0xF0, 0xF8, 0x08, 0x88, 0xC8, 0xF8, 0xF0, 0x00, 0x0F, 0x1F, 0x13, 0x11, 0x10, 0x1F, 0x0F},
	{0x00, 0x20, 0x30, 0xF8, 0xF8, 0x00, 0x00, 0x00, 0x00, 0x10, 0x10, 0x1F, 0x1F, 0x10, 0x10},
	{0x30, 0x38, 0x08, 0x08, 0x88, 0xF8, 0x70, 0x00, 0x18, 0x1C, 0x16, 0x13, 0x11, 0x10, 0x10},
	{0x30, 0x38, 0x88, 0x88, 0x88, 0xF8, 0x70, 0x00, 0x0C, 0x1C, 0x10, 0x10, 0x10, 0x1F, 0x0F},
	{0x00, 0x80, 0xC0, 0x60, 0x30, 0xF8, 0xF8, 0x00, 0x03, 0x03, 0x02, 0x02, 0x02, 0x1F, 0x1F},
	{0xF8, 0xF8, 0x88, 0x88, 0x88, 0x88, 0x08, 0x00, 0x08, 0x18, 0x10, 0x10, 0x10, 0x1F, 0x0F},
	{0xE0, 0xF0, 0x98, 0x88, 0x88, 0x88, 0x00, 0x00, 0x0F, 0x1F, 0x10, 0x10, 0x10, 0x1F, 0x0F},
	{0x08, 0x08, 0x08, 0x08, 0xC8, 0xF8, 0x38, 0x00, 0x00, 0x00, 0x1C, 0x1F, 0x03, 0x00, 0x00},
	{0x70, 0xF8, 0x88, 0x88, 0x88, 0xF8, 0x70, 0x00, 0x0F, 0x1F, 0x10, 0x10, 0x10, 0x1F, 0x0F},
	{0xF0, 0xF8, 0x08, 0x08, 0x08, 0xF8, 0xF0, 0x00, 0x00, 0x11, 0x11, 0x11, 0x19, 0x0F, 0x07},
	{0x00, 0x00, 0x00, 0x60, 0x60, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x0C, 0x0C, 0x00, 0x00},
	{0x00, 0x00, 0x00, 0x60, 0x60, 0x00, 0x00, 0x00, 0x00, 0x00, 0x10, 0x1C, 0x0C, 0x00, 0x00},
	{0x00, 0x00, 0x80, 0xC0, 0x60, 0x30, 0x10, 0x00, 0x00, 0x01, 0x03, 0x06, 0x0C, 0x18, 0x10},
	{0x80, 0x80, 0x80, 0x80, 0x80, 0x80, 0x80, 0x00, 0x04, 0x04, 0x04, 0x04, 0x04, 0x04, 0x04},
	{0x00, 0x10, 0x30, 0x60, 0xC0, 0x80, 0x00, 0x00, 0x00, 0x10, 0x18, 0x0C, 0x06, 0x03, 0x01},
	{0x70, 0x78, 0x08, 0x08, 0x88, 0xF8, 0x70, 0x00, 0x00, 0x00, 0x00, 0x1B, 0x1B, 0x00, 0x00},
	{0xE0, 0xF0, 0x10, 0x90, 0x90, 0xF0, 0xE0, 0x00, 0x0F, 0x1F, 0x10, 0x17, 0x17, 0x17, 0x03},
	{0xF0, 0xF8, 0x08, 0x08, 0x08, 0xF8, 0xF0, 0x00, 0x1F, 0x1F, 0x01, 0x01, 0x01, 0x1F, 0x1F},
	{0xF8, 0xF8, 0x88, 0x88, 0x88, 0xF8, 0x70, 0x00, 0x1F, 0x1F, 0x10, 0x10, 0x10, 0x1F, 0x0F},
	{0xF0, 0xF8, 0x08, 0x08, 0x08, 0x38, 0x30, 0x00, 0x0F, 0x1F, 0x10, 0x10, 0x10, 0x1C, 0x0C},
	{0xF8, 0xF8, 0x08, 0x08, 0x18, 0xF0, 0xE0, 0x00, 0x1F, 0x1F, 0x10, 0x10, 0x18, 0x0F, 0x07},
	{0xF8, 0xF8, 0x88, 0x88, 0x88, 0x08, 0x08, 0x00, 0x1F, 0x1F, 0x10, 0x10, 0x10, 0x10, 0x10},
	{0xF8, 0xF8, 0x88, 0x88, 0x88, 0x08, 0x08, 0x00, 0x1F, 0x1F, 0x00, 0x00, 0x00, 0x00, 0x00},
	{0xF0, 0xF8, 0x08, 0x08, 0x08, 0x38, 0x30, 0x00, 0x0F, 0x1F, 0x10, 0x11, 0x11, 0x1F, 0x0F},
	{0xF8, 0xF8, 0x80, 0x80, 0x80, 0xF8, 0xF8, 0x00, 0x1F, 0x1F, 0x00, 0x00, 0x00, 0x1F, 0x1F},
	{0x00, 0x00, 0x08, 0xF8, 0xF8, 0x08, 0x00, 0x00, 0x00, 0x00, 0x10, 0x1F, 0x1F, 0x10, 0x00},
	{0x00, 0x00, 0x00, 0x08, 0xF8, 0xF8, 0x08, 0x00, 0x0C, 0x1C, 0x10, 0x10, 0x1F, 0x0F, 0x00},
	{0xF8, 0xF8, 0xC0, 0x60, 0x30, 0x18, 0x08, 0x00, 0x1F, 0x1F, 0x03, 0x06, 0x0C, 0x18, 0x10},
	{0xF8, 0xF8, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x1F, 0x1F, 0x10, 0x10, 0x10, 0x10, 0x10},
	{0xF8, 0xF8, 0x30, 0xE0, 0x30, 0xF8, 0xF8, 0x00, 0x1F, 0x1F, 0x00, 0x00, 0x00, 0x1F, 0x1F},
	{0xF8, 0xF8, 0xC0, 0x80, 0x00, 0xF8, 0xF8, 0x00, 0x1F, 0x1F, 0x00, 0x01, 0x03, 0x1F, 0x1F},
	{0xF0, 0xF8, 0x08, 0x08, 0x08, 0xF8, 0xF0, 0x00, 0x0F, 0x1F, 0x10, 0x10, 0x10, 0x1F, 0x0F},
	{0xF8, 0xF8, 0x08, 0x08, 0x08, 0xF8, 0xF0, 0x00, 0x1F, 0x1F, 0x01, 0x01, 0x01, 0x01, 0x00},
[... 439 lines cut ...]
	{0x7F, 0x7F, 0x04, 0x04, 0x7C, 0x78},
	{0x00, 0x00, 0x7D, 0x7D, 0x00, 0x00},
	{0x20, 0x60, 0x40, 0x7D, 0x3D, 0x00},
	{0x7F, 0x7F, 0x18, 0x3C, 0x66, 0x42},
[... cut ...]
	{0x7E, 0x7E, 0x02, 0x06, 0x04, 0x00},
	{0x44, 0x4E, 0x4A, 0x4A, 0x7A, 0x30},
	{0x3F, 0x7F, 0x44, 0x44, 0x40, 0x00},
	{0x3E, 0x7E, 0x40, 0x40, 0x7E, 0x3E},
	{0x0E, 0x3E, 0x60, 0x60, 0x3E, 0x0E},
	{0x3E, 0x7E, 0x30, 0x30, 0x7E, 0x3E},
	{0x66, 0x7E, 0x18, 0x18, 0x7E, 0x66},
	{0x0E, 0x5E, 0x50, 0x50, 0x7E, 0x3E},
	{0x62, 0x72, 0x5A, 0x4E, 0x46, 0x00},
	{0x08, 0x3E, 0x77, 0x41, 0x00, 0x00},
	{0x00, 0x00, 0x7F, 0x00, 0x00, 0x00},
	{0x00, 0x41, 0x77, 0x3E, 0x08, 0x00},
	{0x0C, 0x06, 0x0C, 0x18, 0x0C, 0x00}
	};
#endif`

PieeerB, 2023-10-18:
If you replace VFO with VF, then you can fit four frequency signs, here is an example
[image]

egzumer, 2024-01-15:
I thought about adding a compile option to switch between fonts, but coming back to this I don't see any difference in this font.

closed 2024-01-15 by egzumer