egzumer/uv-k5-firmware-custom #34: AIRBAND AM Problem in V.012

hssbo, 2023-10-18:
Hello EGZUMER! Congratulations on your work, I really appreciate your firmwares. In v0.12 I noticed a loss of sensitivity in AM airband reception. I live near an airport and after this update reception decreased a lot. I hope I'm helping with my feedback! TKS

hssbo, 2023-10-18:
Yes, I downgraded to v0.11...

egzumer, 2023-10-18:
Did you try back and forth and it is always better on 11 and much worse on 12? As I said nothing was changed whatsoever that could cause that. Also I don't notice any degradation in AM reception.

hssbo, 2023-10-18:
Yes, is always better on 11...
Reception is more sensitive in version 11... the number of listens is greater...
Version 12 seems "deaf" to airband...

JMSCompSci, 2023-10-18:
Did you enable the "AM Fix" option in settings?  It boosted the audio big time for me.

egzumer, 2023-10-19:
I've set up two radios, one with v11 and second with v12 and run them together. There is no difference. Both at the same time on sql 4 receive even as low signals that are noisy and hard to understand.
I also asked on my local chat, nobody confirmed this problems.

hssbo, 2023-10-19:
Anyway, maybe it was a false impression I had. Thank you very much for your attention!

hssbo, 2023-10-19:
> Did you enable the "AM Fix" option in settings? It boosted the audio big time for me.

Yes

closed 2023-10-19 by egzumer