egzumer/uv-k5-firmware-custom #36: Spectrum freezes when switch to AM

Skelm01, 2023-10-18:
Hello,
I inform you that Spectrum keep freezing when I switch to AM. 
For example: VFO 135.450mhz - F5 - 0

egzumer, 2023-10-19:
Every time? This was reported before #4 but I can't reproduce that, never happened to me.

egzumer, 2023-10-19:
Use [k5prog_win.exe](https://github.com/OneOfEleven/k5prog-win/raw/main/k5prog_win.exe) to dump config and calibration, and share the binaries here. This has to be related to particular configuration I think. Maybe I'll be able to to trigger this on my radio with your settings. I don't have a clue otherwise.

Skelm01, 2023-10-19:
> Use [k5prog_win.exe](https://github.com/OneOfEleven/k5prog-win/raw/main/k5prog_win.exe) to dump config and calibration, and share the binaries here. This has to be related to particular [...]

Ok, I got the 2 config files, how I can I share here?

Skelm01, 2023-10-19:
> Use [k5prog_win.exe](https://github.com/OneOfEleven/k5prog-win/raw/main/k5prog_win.exe) to dump config and calibration, and share the binaries here. This has to be related to particular [...]

Bin files are not supported for sharing, how can I do?

egzumer, 2023-10-19:
Zip them. Also write what steps are needed to reproduce. Turn on the radio, F+5 and wait?

Skelm01, 2023-10-19:
> Zip them. Also write what steps are needed to reproduce. Turn on the radio, F+5 and wait?
VFO on Air band AM, F+5 wait 2-3 sec then press 0 to activate AM

[my_config.zip](https://github.com/egzumer/uv-k5-firmware-custom/files/13047140/my_config.zip)

egzumer, 2023-10-20:
I didn't manage to reproduce the problem unfortunately.

closed 2023-12-27 by egzumer