egzumer/uv-k5-firmware-custom #39: side buttons menu add new function

CaptainRi, 2023-10-21:
Among many firmware versions, I really like your firmware version and it has become perfect. Thank you very much for your selfless efforts！

**Can you add the function of 2nd PTT to the F1shrt or F2shrt options to transmit the second frequency, so that those in need can freely choose from multiple functions，that you can choose monitor，others can choose the functions they need.**

If this leads to excessive firmware volume, options can be added to the makefile to allow users to decide whether to add features.

egzumer, 2023-10-21:
I tried to add that, but the code is so messy and tied up, that I gave up for now.

CaptainRi, 2023-10-21:
> 我试图添加这一点，但代码是如此混乱和束缚，所以我暂时放弃了。

Okay, don't worry

closed 2023-10-21 by CaptainRi