egzumer/uv-k5-firmware-custom #44: Fixed restoring channel number/frequency on scan exit

Wujamichu, 2023-10-23:
Hello 
In the version v0.12 you changed this  - Fixed restoring channel number/frequency on scan exit

is it possible to return in the new software so that after  exit of  scaning,  it does not return to the last channel before scanning, but stays on the channel where the scan stopped, this was a good solution when the radio found a active station, then after pressing the exit it stayed on this frequency and now returns to channel that was set before the scanning has started. 
If it won't be changed, can I change it myself? Where can i find this ? 
Best Regards 
Pawel

Wujamichu, 2023-10-24:
welcome back
I tried with PTT but it returns to the channel from which it starts scanning, but I found a solution - when scanning stops, press the menu first, then exit and it stays on the channel where scanning stopped.

PieeerB, 2023-10-24:
He wants to stop scanning on the channel regardless of whether there was a signal or not, stopping scanning on the channel from which the scan was started does not make any sense. You may just not have time to press the PTT button and the channel will return to the beginning, and if it stopped where you stopped it, it would be easier to find the missed signal.

egzumer, 2023-10-24:
> He wants to stop scanning on the channel regardless of whether there was a signal or not, stopping scanning on the channel from which the scan was started does not make any sense.

On which channel? What is the point of stopping on some channel if there was no signal? The scanner scans over 10 channels a secondmind you.
On the other hand if the signal is found, you have couple of seconds to react.

egzumer, 2023-10-25:
Prerelease if you want to test.
# Changes:
* Fixed offset input menu not working properly
* New configurable button function: `VFO/MR`
* Changed airband upper limit from 136MHz to 137MHz
* `MENU` button now saves frequency/channel scan result (same as `PTT`, `EXIT` still discards the search and goes back to initial channel/frequency)
* Scanner now remembers last found frequency/channel and goes to it when scan result is accepted (`PTT` or `MENU` as stated above)

[firmware.packed.bin.zip](https://github.com/egzumer/uv-k5-firmware-custom/files/13146781/firmware.packed.bin.zip)