egzumer/uv-k5-firmware-custom #45: rssi/mic bar is hard to read

QuickXand, 2023-10-24:
the rssi bar is too hard to reach the end. why not using the more common style like this?
[image]

CaptainRi, 2023-10-24:
I have been hesitant to ask the same question. RSSI S bar can be optimized to look better, such as the style of zebra stripes, Or a style similar to YAESU FT-4XR.

hssbo, 2023-10-25:
This RSSI bar is very beautiful...

ErikS-web, 2023-10-25:
Is this what you prefer ? see here https://t.me/UVK5FW/1148
[image]

hssbo, 2023-10-25:
> Is this what you prefer ? see here https://t.me/UVK5FW/1148 [image]
> 
> Just testing and seems to work fine.

I installed it and it's working fine! I just disabled the rssi bar microphone so there wouldn't be different bars. TKS!

egzumer, 2023-10-26:
Implemented on 38b1e5cbaa669f75f198da0f2cd1466408387706 and released

closed 2023-10-26 by egzumer

QuickXand, 2023-10-28:
> Implemented on [38b1e5c](https://github.com/egzumer/uv-k5-firmware-custom/commit/38b1e5cbaa669f75f198da0f2cd1466408387706) and released
Thanks and very brillant job!
Is there any possible to make mic indicator looks same as the s-bar?

CaptainRi, 2023-10-29:
> > Implemented on [38b1e5c](https://github.com/egzumer/uv-k5-firmware-custom/commit/38b1e5cbaa669f75f198da0f2cd1466408387706) and released
> > Thanks and very brillant job!
> > Is there any possible to make mic indicator looks same as the s-bar?

@egzumer