egzumer/uv-k5-firmware-custom #49: Set the RX backlight for channel A or B separately.

troilus, 2023-10-25:
Is it possible to set the RX backlight for channel A or B separately to prompt the user which channel channel is receiving.

ErikS-web, 2023-10-25:
you have **RX** on screen when signal is recieved, in front of specific channelname/frequency on A- or B-channel

troilus, 2023-10-25:
> you have **RX** on screen when signal is recieved, in front of specific channelname/frequency on A- or B-channel

Yes, I know, but in some scenarios, such as driving, it is really not that convenient and safe to watch the screen.

muromdx, 2023-11-11:
It would be a quite useful enhancement. Small "RX" label is not always clearly visible especially when taking a quick look at the display. For example, UV-R50 marks RX being active last time.

ErikS-web, 2023-11-14:
I think the backlight is on or off. I haven't read about "Split" backlight anywhere yet.
But have you ever tried the RX mode in **DUAL RX RESPOND**?
I have copied some tekst from the [Wiki-Menu](https://github.com/egzumer/uv-k5-firmware-custom/wiki/Menu) 

59. `RxMode` - sets how the upper and lower frequency is used
   * MAIN ONLY - always transmits and listens on the main frequency (NO extra characters on screen)
   * DUAL RX RESPOND - listens to both frequencies, if signal is received on the secondary frequency it locks to it for a couple of seconds so you can respond to the call (**`DWR`** on screen)
   * CROSS BAND - always transmits on the MAIN/primary and listens on the secondary frequency (**`XB`** on screen)
   * MAIN TX DUAL RX - always transmits on the primary, listens to both (**`DW`** on screen)
>    Explanation of how these modes function
>    * The main channel is marked `►` and with PTT active `►TX`
>    * A receiving channel is marked `RX` as soon as a signal is received. Other VFO/channel is blocked from RX.
>    * With DUAL RX RESPOND, the secondary VFO/channel is marked `>` as temporarily the main channel when something is received there. 
>    * If nothing is received after a timer of 4 seconds (`><` on screen), that status will expire.

But when driving a car **MAIN ONLY** might be a saver setting. !

closed 2023-11-20 by troilus