egzumer/uv-k5-firmware-custom #50: AM reception

Fabuseta, 2023-10-25:
Hi,
First I have to thank you for the best all-around firmware for these radios. Keep doing the great job! 

And about the problem:

Findings from the current version of your software (egzumer-UV-K5 1of11+fagci_ 25.10.23v1.packed.bin). I noticed a change in AM reception regarding aviation frequencies. The sound is quieter, and there are noises that have not appeared in previous software versions. Kind of windy, swirly(?) sound during reception, that I have no heard before. And also I have to adjust the volume all the way up so that I can hear anything properly at all. On the other hand, now I can hear weaker signals more clearly (e.g. -90dbi - -100dbi), which I haven't heard properly before. I switched back to the previous software version (egzumer-UV-K5 1of11+fagci_ 22.10.23v1.packed), and the AM reception went back to "normal".

Video link for the new firmware reception: https://www.dropbox.com/scl/fi/f3o5ls5kcfuurzmyzlb9n/VID_20231025160119.mp4?rlkey=tkftphkzss5b9clnu8wsheh9g&dl=0

Video link for the "normal" reception: https://www.dropbox.com/scl/fi/vl9hcdvklz1v3uhyimixn/VID_20231003165949.mp4?rlkey=5tk6sytc2xqdzjz5othsld20w&dl=0

The one with the "normal" reception, was with quite an old firmware (with the AM fix on), However It compares qualitatively with other older software when it comes to AM reception.
Firmware bins that I have been using are downloaded from the tha Quansheng UV-K5 Firmwares -Telegram channel.

egzumer, 2023-10-25:
These are not my releases so I don't know what version it is, but probably it is built when AGC was turned on on the main branch, (it is not anymore). The AGC is experimental, was copied from 1o11 but after tests I decided to pull it out.

Fabuseta, 2023-10-26:
These were your custom firmwares, but builds were compiled by Lothe Shar. 
And you are probably right about that AGC thing. It was also discussed elsewhere that it did not bring the desired improvement in sound quality. But anyway, now it's sounding again great!
Thanks again for your work! :)

closed 2023-10-26 by egzumer