egzumer/uv-k5-firmware-custom #53: New Smeter issue when Squelch set to 0

Skelm01, 2023-10-26:
I like new Smeter looks, but I noticed that when the squelch is set to 0 or pressing the side monitor button, the smeter shows a S8, instead do two or three lines more consistent with background noise

rai4424, 2023-10-27:
I hink its not a major isue.

egzumer, 2023-10-27:
The new s-meter is only a visual change, all the underlaying values are all the same. It all comes from the RSSI which is being read directly from BK4819.
dBm value is calculated as `(rssi / 2) - 160` (don't know where it comes from), and the resulting value is used to set the s-meter as described [here](https://wikipedia.org/wiki/S-meter)
To be clear, the underlaying code comes from 1o11, I'm not a radio amateur, I learn as I go, so I don't know many things, but..
It all seems wrong, the original s-meter in stock firmware is being calibrated at the factory and written to EEPROM. 4 levels are being calibrated for different values of RSSI. Each band 1-7 is calibrated separately.
Now the middle screen s-meter doesn't use calibration at all so it cannot be considered reliable I would say. I think the calibration needs to be used but how to use those 4 points to infer dBm that I don't know yet.

egzumer, 2023-10-27:
It turns out the RSSI levels are not calibrated at factory. They are written to eeprom but are the same on each radio. But the levels are different for 2 groups of bands. This was not taken into account in the s-meter

bands 3,4,5,6,7
lvl 1: 110
lvl 2: 120
lvl 3: 130
lvl 4: 140
bands 1,2
lvl 1: 180
lvl 2: 190
lvl 3: 200
lvl 4: 210

egzumer, 2023-10-27:
I did change the algorithm for s-meter. I don't have many transmission around me but it seems that it works much better.
It takes first and last calibration points, it assumes it is S1 and S9 and does linear approximation from that.
Let me know how it works for you.
[firmware.packed.bin.zip](https://github.com/egzumer/uv-k5-firmware-custom/files/13191573/firmware.packed.bin.zip)

TokeTokerson, 2023-10-28:
https://github.com/egzumer/uv-k5-firmware-custom/assets/148666132/9652e7ff-029e-4aec-addd-259a941f9773
On 162.400 MHz it seems to have went tighter. Prior to this update (from the link you provided above), this NOAA signal was -90db with an S9 on the bar. This seems to have made it far less sensitive (at least what's being reflected on the display / meter.)

TokeTokerson, 2023-10-28:
Here's the same frequency on version .14
https://github.com/egzumer/uv-k5-firmware-custom/assets/148666132/8b48fa14-d942-4e27-8bef-06b7afabba1a

egzumer, 2023-10-28:
As I said I'm not a HAM operator so I'm asking about opinion. Is it better or is it worse? Tighter, less sensitive means nothing.
The real solution would be if someone did measure full spectrum characteristic with a frequency generator or something.

egzumer, 2023-10-28:
I couldn't see what the links are on the phone.
So now it says S0 like there would be no signal. So not too good, but this is on one frequency. I get better results on other frequencies, but it gets overdriven for high strength signals, when I transmit with the other radio it goes up to 200dBm :).
I have no reference on what it should be so this makes no sense unless someone do some measurements or comparison with other decent radio on multiple frequencies.

This version shows raw RSSI value. I need dBm values for different frequencies and different signal strengths
[rssi.zip](https://github.com/egzumer/uv-k5-firmware-custom/files/13194917/rssi.zip)

crazy4volvo, 2023-10-28:
Maybe this article will be helpful: [https://www.giangrandi.org/electronics/radio/smeter.shtml](url)

egzumer, 2023-10-28:
This is the same as wikipedia page I gave in 3rd post. I know this, the problem is what is the relation of RSSI to dBm. I don't have 50R load nor RF voltage meter.

wmarkow, 2023-11-10:
Related to this:
```
dBm value is calculated as (rssi / 2) - 160 (don't know where it comes from)
```

```
the problem is what is the relation of RSSI to dBm
```

Take a look at [BK4819(V3) Application Note 20210428.pdf](https://github.com/amnemonic/Quansheng_UV-K5_Firmware/blob/main/hardware/BK4819(V3)%20Application%20Note%2020210428.pdf), page 13. The equation from first quotation seems to be correct. The pdf says nothing more about. However recently I was playing around with a SI44xx chip and I have found something interesting in their [Si4464-63-61-60.pdf](https://github.com/wmarkow/arduino-sandbox/blob/master/fox-hunting-stm8/doc/Si4xxx/Si4464-63-61-60.pdf), page 31, the same equation is mentioned with a small change:
```
The RSSI value read by the API could be translated to dBm by the following linear equation:
RSSI (in dBm) = (RSSI_value /2) – RSSIcal
RSSIcal in the above formula depends on the matching network, modem settings, and external LNA gain (if
present). The RSSIcal value can be obtained by a simple calibration with a signal generator connected at the
antenna input. Without external LNA, the value of RSSIcal is around 130 ±30.
```

Does it put a bit of light to the case?

muromdx, 2023-11-12:
4pda's user says that there is correct S-meter in 15th and 17th releases of UV_K5_playground firmware. I have tested custom FW based on it and I can say that S-meter works properly. I'm not a programmer myself. Is it possible to get a part of code responsible for S-meter from that firmware?

egzumer, 2023-11-25:
Yes the equation, is correct, it indeed comes from the documentation. The RSSIcal description from SI44xx sheds some light on the problem, however I don't have proper equipment, nor a necessary knowledge. There are some LNAs, I don't know which are external to BK4819.
The S-meter values were adjusted by feel in 67307463e2494b4223d90fb477e28cba76b24d08, which was discussed in #92.

Current build that includes the adjustment:
https://github.com/egzumer/uv-k5-firmware-custom/suites/18492876221/artifacts/1072961377

g4eml, 2023-11-27:
The revised s-meter values are an improvement but still read too high on some bands. 

-130dBm is a typical noise floor for this type of radio and that would be expected to coincide with S0.  Removing the antenna should give readings close to those values on all bands.

I have the necessary test equipment, I will do some measurements on each band and test the firmware with suitable adjustments.  I will then post the results here.  There will be some variation between radios but I suspect they will not be that significant. 

Colin G4EML

egzumer, 2023-11-28:
Why S0 changed to -130?
Sources I read mention -141 for S1 for frequencies above 30 MHz. S0 is presumed to be -6dBm lower hence it was -147.

g4eml, 2023-11-29:
Those sources are probably talking about narrow bandwidths such as used for CW and SSB. FM and AM radios have much wider filters. The noise floor increases with increasing bandwidth. -130 dBm noise floor is fairly typical for an FM radio. S values are not absolute measurements. They should always be adjusted relative to the noise floor which is the reading with no signal. Removing the antenna is the best way to ensure there is no signal. In that condition the S meter should read S0. Using a base level of -147 would mean the lowest the S-Meter would ever read would be +17dB or just below S3. A reading of S3 with no signal present makes no sense.

egzumer, 2023-11-29:
Thank you for the measurements and for explanation. I pushed those changes, will be included in the next release.

TonySlater, 2023-12-01:
Is there a possibility to add a VOX DELAY option to the menu?  I am looking to use this radio with VOX and such an adjustment would be really useful.

brookbonde, 2023-12-01:
Great work with the firmware!
Is there a possibility to add a VOX DELAY option to the menu? I am looking to use this radio with VOX and a delay adjustment would be really useful.  Thank you.

brookbonde, 2023-12-02:
Can I make a few suggestions for additions to the firmware, if they could be implemented.

Add:
VOX DELAY
STARTUP ON FM RADIO (VHF FM) default boot option
S-Meter on FM RADIO
MIC +20db (some speaker mics slightly require more gain)

Thanks Everyone

egzumer, 2023-12-02:
@g4eml, you guys talk to each other in #174

samhiatt, 2024-11-08:
> As I said I'm not a HAM operator so I'm asking about opinion. ...

> I don't have 50R load nor RF voltage meter.

😲 Wait, what?! Your name is now synonymous with the leading firmware for the most hackable ham radio... and you're not a ham? You don't even have an RF voltage meter? 🤯 I think we need to pull together to buy you one, or something. The community is in your debt.

egzumer, 2024-11-08:
> The community is in your debt.

Thank you, I'm glad my work is usefull for someone, but there is no debt. I based the project on someone elses work, there are some other projects that are based on my work. That is the magick of free and open source software. I did my part, unfortunately I haven't had time to continue the work, the project is stalled for now.