egzumer/uv-k5-firmware-custom #54: Cutting in Rx (v.14)

rai4424, 2023-10-27:
Problem in Strong signal reception .
Cutting in Rx in v.14.
(Signal -47 +20, cutting started when receiving audio)

egzumer, 2023-10-27:
Try transmitting an any frequency, it resets some register that is set only once on reboot. It should fix the issue. This is a change that I unknowingly copied from 1o11. Let me know if this is it. I'll also release a fix shortly.

rai4424, 2023-10-27:
> Try transmitting on any frequency, it resets some register that is set only once on reboot. It should fix the issue. This is a change that I blindly copied from 1o11. Let me know if this is it. I'll also release a fix shortly.

 Audio and signal reception of this Fw is awesome, investigation of this may be helpful to remove this problem.

/Custom_Firmware_by_Matoz_V0.29C_fagci_keys_style.bin

closed 2023-10-27 by egzumer