egzumer/uv-k5-firmware-custom #55: D Decd error

CaptainRi, 2023-10-27:
Two k5s, A is the original firmware, B is v0.14.1 firmware.
both same freq eg438.500, both with DTMF (D Decd) turned **on**. After this setting, B calls A with only light and no sound, 
but both with DTMF (D Decod) turned **off** become normal

closed 2023-10-29 by egzumer

CaptainRi, 2023-10-30:
> DTMF decoding is originally meant to give an option to call each other somewhat like with phones. The call pattern is Where callee and caller are 3 digit ANI ID of each radio. If radio receives a call it only then starts to listen for a preset amount of time. This was broken in 1o11's repo when I forked it. I will try to fix that in the future.`callee*caller`

same issues still exists

egzumer, 2023-10-30:
You didn't actually say what is the problem, you only described what is going on. So what is the actual problem?

CaptainRi, 2023-10-30:
> 你实际上并没有说出问题是什么，你只是描述了发生了什么。那么实际问题是什么？

Yes, I only know the phenomenon of the problem. I don't understand programming, and of course I don't know where the problem lies

CaptainRi, 2023-10-30:
> You didn't actually say what is the problem, you only described what is going on. So what is the actual problem?

Yes, I only know the phenomenon of the problem. I don't understand programming, and of course I don't know where the problem lies

egzumer, 2023-10-30:
No, not where the problem lies, I will investigate where it lies, but you didn't state the problem at all.

CaptainRi, 2023-10-30:
> No, not where the problem lies, I will investigate where it lies, but you didn't state the problem at all.

I told you the method of reproduction. What's not clear about it

egzumer, 2023-10-30:
You only described what you do, you didn't state what is not right in the result.

egzumer, 2023-10-30:
I fixed what I consider wrong, but you say it is still the same so I'm getting confused

CaptainRi, 2023-10-30:
> No, not where the problem lies, I will investigate where it lies, but you didn't state the problem at all.

Two k5s, A is the quansheng original firmware, B is your firmware today.
both same freq （eg438.500）, both with DTMF (D Decd) turned on. After this setting
Result B calls A , A only light   No sound,
A calls B，B only light No sound

CaptainRi, 2023-10-30:
> I fixed what I consider wrong, but you say it is still the same so I'm getting confused

Wait a few minutes, I will send the video to the telegram

egzumer, 2023-10-30:
Can you see you see any "bad", "wrong", "not right", "error", "not expected" in your statement? Because I don't. How am I supposed to ask the question in any other way?

You do certain actions, and you get certain outcome. Can you state what is different in what is going on, that is different from what you expect to happen?

egzumer, 2023-10-30:
Radio with `D Decd` set to ON should have no sound on receive, unless you call it with its ANI ID. When I call my radio with 102*103 I get the sound.

https://github.com/egzumer/uv-k5-firmware-custom/wiki/Radio-operation#dtmf-calling-decoding

CaptainRi, 2023-10-30:
> Radio with `D Decd` set to ON should have no sound on receive, unless you call it with its ANI ID. When I call my radio with 102*103 I get the sound.
> 
> https://github.com/egzumer/uv-k5-firmware-custom/wiki/Radio-operation#dtmf-calling-decoding

I understand. I'm very sorry for wasting a lot of your energy and time.Thank you very much for everything you selflessly did for K5！

commit f7c25b48 (2023-10-29): FIX #55: Broken DTMF calling (D Decd)