egzumer/uv-k5-firmware-custom #56: F+7 to change  AM\FM\SSB

troilus, 2023-10-27:
Would you please add F+7 to change  AM\FM\SSB like fagci mod? And display AM\FM\SSB in VFO or MR.

Skelm01, 2023-10-27:
With this enhancement your FW should be almost perfect

ErikS-web, 2023-10-28:
Or make this under Long-Press-7 ....

egzumer, 2023-11-02:
3705b7c2c06c5936c1089bf54d5a714b304967be SSB demodulator
ede1edd905ed5d13cd576ee3a758692b07e27b8e Additional function for configurable buttons that switch demodulation modes

The switching modes isn't something used by most people so I'm not keen on setting it permanently for any button, programmable function should suffice.

egzumer, 2023-11-02:
If you could confirm that it all works as it should, that would be nice.
[ssb.packed.bin.zip](https://github.com/egzumer/uv-k5-firmware-custom/files/13242807/ssb.packed.bin.zip)

DinoNik, 2024-02-29:
Hello Egzumer,
just one question. I use your firmware version 0.22 and I unlock all but in USB mode I have only RX no TX  when I press PTT button I get message TX Disable. Is it possible for version 0.22 to work on SSB and if yes what I do wrong, thx

egzumer, 2024-02-29:
You can tx with FM only.

BorisTeskera2019, 2024-02-29:
Hello to all,
I try to insert new firmware in ma Quansheng UV 5R and I get message "The length of file is to much".
Any tips, advice, thanks.
[image]