egzumer/uv-k5-firmware-custom #59: Modulation is low

RoadRangerX, 2023-10-28:
Love 0.14.1 !
I have the mic gain up full and have the TX compandor on (not convinced that helps?) but am still maybe 6dB shy of having decent modulation.
My Retevis RT-3S with the OpenGD77 firmware has 33dB of gain available past where it's about perfect  :-O
Is it possible to get more gain from the firmware?
Thanks!

ErikS-web, 2023-10-28:
Retevis RT-3S is an other type of hardware compared to UV-K*.

egzumer, 2023-10-28:
From what I can see in the code the Mic setting in the menu gives all that is possible.

RoadRangerX, 2023-10-28:
I now read elsewhere that the Compandor function is broken - in fact I think it adds static and not gain. Maybe when that is fixed it will help. Or maybe I'll sus out the code and find a way to add some audio processing :)

egzumer, 2023-10-28:
The compander is meant to be used on both radios. Transmitter compresses audio receiver expands it. If only one uses it the audio would be distorted. Also the effect is minimal.

RoadRangerX, 2023-10-28:
I've recorded from my local repeater and do see the Tx Compandor working and doing something useful even without the Rx part.
The three sections of this graph show first another operator, then me with gain max and Tx Compandor on, then with the Compandor off.  It shows that I need about another 12dB. The Tx Compandor does add a slight amount of gain and does limit the peaks somewhat.
[image]

RoadRangerX, 2023-10-28:
Same graph with the last two transmissions amplified by 12 dB (in Audacity)
[image]

closed 2023-10-30 by egzumer