egzumer/uv-k5-firmware-custom #60: Detect twice EXIT and frefresh screen to solve screen rubbisch after HF interference.

ErikS-web, 2023-10-28:
This [enhancement] might be a solution for UV-K5(8) where the orange backlighted screen is sensitive for HF.
This sometimes rusults in stripes on screen or blanc screen.
Just place hand-palm on screen, with fingers near antenna, and transmit. And screen becomes rubbisch.

Solution (hardware) power-of and on which will reset the porto and thus the screen inteface.

On telegrams suggestion was done for sw-reinitilisation in stead of hw-reset.
My suggestion is to "detect twice EXIT- button pressed" and then "reinitialise screen" or "sw-reset porto". 
Whatever is possible to realise.

egzumer, 2023-10-28:
I have all 3 QS models, I tested it just now, none of them have that problem. 9 QS radios went through my hands, never experienced this issue. It doesn't seem to be worth spending the time to fix that in code. Just don't cover the screen with your palm while transmitting.

ErikS-web, 2023-10-28:
Well I've seens it several times mentioned om forums and Telegram from different users, and with foto's. 
I can reproduce this problem. But it's not an urgent problem. 
Suggest a prio-3. Wintertime will come ;-)

egzumer, 2023-10-28:
Even if, I still can't reproduce the problem, so I can't test and see what helps. I wouldn't be shocked if software reset would be impossible. It could be hardware hanging.

IT9-TOP, 2023-10-28:
I have 3 k5s, 2 5R_plus and 1 k5 and the problem occurs on all of them, and many users have this problem

sq2mo, 2023-11-02:
Also experiencing the issue, on two different models - K5 and K5(8). Factory FW does not have this issue. 

Earlier versions of OneOfEleven FW didn't have the problem, so I did a small bisection and tracked this down to a certain commit from 21st Sept: 
https://github.com/OneOfEleven/uv-k5-firmware-custom/commit/835378c78ca1064aa08e6ff5528eddbb9ac4e900

It seems that RFI is interfering with display I2C communication when display contents are changed. Indeed, it's a hardware thing, but perhaps it could be worked around in SW.

Same issue:
https://github.com/OneOfEleven/uv-k5-firmware-custom/issues/71
https://github.com/OneOfEleven/uv-k5-firmware-custom/issues/39

closed 2023-11-03 by egzumer

sq2mo, 2023-11-03:
WA works. Thank you for your work!

ErikS-web, 2023-11-05:
Solved with "screen initialization" after each PTT-release. in [`Release 0.16`](
https://github.com/egzumer/uv-k5-firmware-custom/releases/tag/v0.16)

commit 72fc4bf5 (2023-11-03): FIX #60: LCD interference glitch