egzumer/uv-k5-firmware-custom #61: Add real MDC1200 support

QuickXand, 2023-10-28:
The latest 1o17 mod supported mdc encoding. Is it possible to add this feature?

egzumer, 2023-10-28:
It is not attractive to me, I don't see a point

CaptainRi, 2023-10-29:
OneOfEleven commented 19 hours ago
The problem (a big one) with doing MDC receive is the the K5 radio chip requires a short length of bit toggling pre-amble at the start of the packet, the pre-amble allows receivers to sync their bit clock PLL's.

But, motorola made a mistake when they created MDC, the designer never added that much needed bit toggling pre-amble, which means the K5 BK4810 FSK modem part can't sync it's bit clock at the start of each and every packet, which means a lot of missed packets.

Most all other packet transmission standards start their packets with this much needed bit toggling pre-amble sequence, motorola MDC don't. It can be made to work IF the FSK modem demodulator is designed to work with MDC, but the K5's BK4819 chip has not been so designed.

The other problem now is CPU flash space.

QuickXand, 2023-10-29:
> It is not attractive to me, I don't see a point

Transceive for mdc is already working on 1o17 firmware and real mdc sounds much better than the builtin fake one

egzumer, 2023-10-29:
> Transceive for mdc is already working on 1o17 firmware and real mdc sounds much better than the builtin fake one

Really? Should I port this to have it sound better? If it's already working there does it mean that it is easy to port here? Try to port it yourself and tell me how it went.

QuickXand, 2023-10-30:
> > Transceive for mdc is already working on 1o17 firmware and real mdc sounds much better than the builtin fake one
> 
> Really? Should I port this to have it sound better? If it's already working there does it mean that it is easy to port here? Try to port it yourself and tell me how it went.

Sorry to make you feel offenced. I'm not meant you should port that,that was only a suggestion.
Your firmware is already perfect for daily use in my opinon.

egzumer, 2023-10-30:
That is fine. Sorry I'm so harsh. You seem to not realize, things like that take hours or even days to finish. It also eats up the small amount of memory space left. What 1o11 did is certenly interesting and shows what the radio can be capable of, but in the end that is a very niche functionality even if it would work perfectly. Spending that much resources on a nonfunctional option is out of the question.

closed 2023-10-30 by egzumer