egzumer/uv-k5-firmware-custom #64: SSB

Big-Mal99, 2023-10-30:
This is the only function that I personally think is missing

troilus, 2023-10-30:
Hi #56

egzumer, 2023-10-30:
I have no idea how to set it up. I have no antenna to receive such signals.

Big-Mal99, 2023-10-30:
Thanks for the prompt and honest reply

egzumer, 2023-11-02:
New modulation and programmable button function for switching modes added. If you can confirm that it works that would be nice.
[ssb.packed.bin.zip](https://github.com/egzumer/uv-k5-firmware-custom/files/13242814/ssb.packed.bin.zip)

Big-Mal99, 2023-11-02:
Thanks for sending me this file.  I have installed it and all seems well
although at the moment I am stuggling to find a station TXing SSB to check
this function out.  I will try again tomorrow when hopefuly I may be able
to find a station.

Malcolm

Big-Mal99, 2023-11-04:
I have had a couple of further looks at this and the problems I am having
with SSB are probably due to my loction.  I assume that v0.16 is the
released version of this test one so I am going to flash that.
Thanks for th great work
Malcolm

egzumer, 2023-11-04:
The release have am fix disabled for ssb, should work better

closed 2023-11-05 by egzumer