egzumer/uv-k5-firmware-custom #66: Not Issue. question about tx

Romashca, 2023-10-30:
is it possible to make a minimum power of about 100-200 MW

egzumer, 2023-10-30:
It is not entirely trivial. All happens around function `FREQUENCY_CalculateOutputPower`. But it uses factory calibration values which are different for each radio. Also there are 3 calibration points for each combination of band and power level. I wouldn't touch without an RF power meter.

egzumer, 2023-10-30:
Didn't you help 1o11 do the power reduction mod?

egzumer, 2023-10-30:
My code is a fork of 1o11. The file where the change is is identical. I pushed the change to the repo. Do you have power meter?

Romashca, 2023-10-30:
yes, of course, what variables can be changed?

egzumer, 2023-10-30:
You have to activate the `ENABLE_REDUCE_LOW_MID_TX_POWER` mod. and the code that it activates divide the calibration values by 5 and by 3. You have to change dividers.

I'm also interested in what values you will get, before the mod, with mod and after your changes. It would be good to do the measurements on the beginning, middle and end of the VHF and UHF band. So whatever results you get please share.

Romashca, 2023-10-30:
[image]not sure where yet

egzumer, 2023-10-30:
Right there, but you didn't pull last changes.

Romashca, 2023-10-30:
frequencyBandTable[Band].lower,
		(frequencyBandTable[Band].lower + frequencyBandTable[Band].upper) / 2,           //  3 or 5?

egzumer, 2023-10-30:
Did you pull the git repository?

Romashca, 2023-10-30:
yes i did

egzumer, 2023-10-30:
I see you didn't, what is the last commit?

Romashca, 2023-10-30:
[image]

egzumer, 2023-10-30:
Now you did, I hope now it's obvious what to change?

egzumer, 2023-10-30:
Well I can see it isn't obvious :)
So not there where you changed it. The dividers between #ifdef and #endif

Romashca, 2023-10-30:
change the number under the fraction , I suppose . in the direction of increase

egzumer, 2023-10-30:
Yes, it already lowers the power by 5 and by 3, if you want to decrease the power, you have to change it to higher numbers. It is repeated 3 times because those are the 3 calibration points for different frequencies on in the band: lower bound, middle and upper

Romashca, 2023-10-30:
DUCE_LOW_MID_TX_POWER  -I D:/Downloads/uv-k5-Egzumer-firmware-custom-main/ -I D:/Downloads/uv-k5-Egzumer-firmware-custom-main//external/CMSIS_5/CMSIS/Core/Include/ -I D:/Downloads/uv-k5-Egzumer-firmware-custom-main//external/CMSIS_5/Device/ARM/ARMCM0/Include -c radio.c -o radio.o
radio.c: In function 'RADIO_ConfigureSquelchAndOutputPower':
radio.c:519:6: error: 'p_vfo' undeclared (first use in this function)
  519 |  if (p_vfo->output_power == OUTPUT_POWER_LOW) {
      |      ^~~~~
radio.c:519:6: note: each undeclared identifier is reported only once for each function it appears in
radio.c:520:3: error: 'TX_power' undeclared (first use in this function)
  520 |   TX_power[0] /= 5;
      |   ^~~~~~~~

egzumer, 2023-10-30:
Ups sorry, 1o11 changed some names and I didn't check that. you have to change p_vfo to pInfo

Romashca, 2023-10-30:
thank you very much

egzumer, 2023-10-30:
There were some other changes, I did push force, if you don't know how to pull that:
> git fetch
git reset --hard origin/master

Romashca, 2023-10-30:
lats try

egzumer, 2023-10-30:
fixed

Romashca, 2023-10-30:
all work  thanks!

Romashca, 2023-10-30:
I will provide you with the data after the measurement

Romashca, 2023-10-31:
measurement data before firmware: H =4.5 , M=3 w ,L=1.2 data after firmware : H=4.5 w , M <1 w, L < 0.5 m .  The measured power in low mode is about 200 MWt, the switch device shows only a slight presence of transmission.  Thank you so much for your help!

closed 2023-10-31 by Romashca