egzumer/uv-k5-firmware-custom #68: expansión de TX en todas las bandas

delta1trieta, 2023-10-31:
Para mi gusto le faltaria una opción en el menú secreto de expansión del TX en todas las bandas. Aun quedan equipos que trabajan en 80 mhz y con este walkie y su correspondiente antena se podria utilizar. Por favor tengalo en cuenta para futuras actualizaciones. Gracias por su gran trabajo.

egzumer, 2023-10-31:
No able espaniol

delta1trieta, 2023-10-31:
For my taste, it would lack an option in the secret menu for TX expansion in all bands. There are still devices that work at 80 MHz and with this walkie and its corresponding antenna it could be used. Please keep this in mind for future updates. Thanks for your great work.

egzumer, 2023-10-31:
Why I'm reluctant to do that is people that didn't do the proper research. As far as I know this radio has negligible transmission power outside VHF and UHF, some microwats on some frequencies, it only transmits in FM and it has huge amount of harmonics that more often than not are transmitted on airband for example.

As an example (close to 80MHz):
[image]

How is that usable?

delta1trieta, 2023-10-31:
I assure you that with the right antenna it is enough to reach the repeaters, I already know the problem with harmonics but to be able to test the equipment they are perfect.
In fact, on a second walkie, I have another firmware installed that does transmit on all bands and I use it for that, but in general it works better with its firmware.

IT9-TOP, 2023-11-01:
I would also like TX on all bands for testing.

DualTachyon, 2023-11-01:
Reminder that IT9-TOP is a toxic person who doesn't think it's normal to provide details on a bug report and loves to waste developers time with non-bugs.

IT9-TOP, 2023-11-01:
You're just good at offending people behind a keyboard, what did I do wrong? Does reporting bugs mean being toxic?
who knows which region you come from, trust a good doctor and get treated.

DualTachyon, 2023-11-01:
You insulted us when we tried to help you. You refused to give information that was requested and demanded we figure it out for you. That's toxic behaviour.

DualTachyon, 2023-11-01:
Not to mention the general bad attitude when you post, and harassing people on telegram.

DualTachyon, 2023-11-01:
You didn't even apologise when we actually solved your problem while you were aggressive to us.

IT9-TOP, 2023-11-01:
It was you who insulted me, don't tell lies, get treatment my friend you have problems.

IT9-TOP, 2023-11-01:
Your behavior is childish, I repeat you have problems.

DualTachyon, 2023-11-01:
Talk trash all you want. All repo owners have been warned about your toxic behaviour.

egzumer, 2023-11-01:
I got the message thank you

Bigpimp347, 2023-11-07:
Although under test's i find 18MHz to 30MHz to be useless, 50MHz and 70MHz show a good frequency and good RF power out, around 4w.
with correct antenna i suspect this would work well.
i also find TX 1296.000 also ok with around 1w of RF.

would it be possible to re instate 50MHz, 70MHz and 1200MHz TX ??

Also although AM and SB Demod is great, the audio is pretty poor and again no TX, 
i would be happy to be given wide TX option but no too bothered if i lose SSB and it has no TX any way.

egzumer, 2023-11-07:
The different modulations are obtain by switching audio output type on the RF IC., and even that is undocumented feature. There is nothing to be configured to be able to change the TX modulation. At least nothing that I know of.

I guess your signal power meter doesn't show what frequency it measures, and if it is fundamental or harmonic?

Bigpimp347, 2023-11-07:
I use a Marconi 2955 and i do have spectrum analysers to check harmonics,
surely the risk is the end user ??
not all owners and users are idiot CB radio users but some are radio amateurs with many years experience in radio, 

i was looking at the radio for us eon 4m and 6m and there is no reason to not fit external band pass filter to clean up the TX signal,
a good 4w at 70MHz and even 50MHz is worth the $20 if all i need is a good microphone (Kenwood make a great mic to fit) and suitable bpf..

egzumer, 2023-11-08:
[tx_for_all.packed.bin.zip](https://github.com/egzumer/uv-k5-firmware-custom/files/13290711/tx_for_all.packed.bin.zip)
Go into hidden menu `F Lock`, choose `UNLOCK ALL`. It wont let you set the option. You have to repeat this action 10 times until it stays on. Do it carefully, if you confirm any other option in the process, the counter will be zeroed, and you will have do it 10 times more.

delta1trieta, 2023-11-08:
Congratulations!!!!!! Now the firmware is great, thank you very much.

delta1trieta, 2023-11-08:
Congratulations!!!!!! Now the firmware is much better but I see that the option (busyCL) stays on (ON) after turning the device off and on even though it is deactivated.
Is it possible to fix that option? Thank you very much in advance.

Bigpimp347, 2023-11-08:
hidden menu F Lock ??
i have been in menu,
but not sure where to find this hidden menu,

trying to find an up to date 'owners manual' also so i can understand some
of the new features..

how do i get the hidden menu ??

egzumer, 2023-11-08:
> option (busyCL) stays on (ON)

It does not stay on. You are trying to set it in a channel mode without saving channel settings.

ErikS-web, 2023-11-08:
See Official release 0.17 under [Hidden Menu option 61](https://github.com/egzumer/uv-k5-firmware-custom/wiki/Menu#hidden-menu)

closed 2023-11-08 by ErikS-web

ChiefRA, 2025-03-28:
Hi @egzumer ,
I've seen that even going into hidden menu F Lock, choose UNLOCK ALL still doesn't unlock the TX on the AM band: (136 - 173 range). 

How can you enable TX on AM band? 
Thanks.

ErikS-web, 2025-03-28:
NOT.

Did you read the [Wiki ](https://github.com/egzumer/uv-k5-firmware-custom/wiki/Radio-operation#tx-on-all-bands)?

ErikS-web, 2025-03-28:
> Hi [@egzumer](https://github.com/egzumer) , I've seen that even going into hidden menu `F Lock`, choose `UNLOCK ALL` still doesn't unlock the TX on the AM band: (136 - 173 range).
> 
> How can you enable TX on AM band? Thanks.

NOT.

Did you read the [Wiki ](https://github.com/egzumer/uv-k5-firmware-custom/wiki/Radio-operation#tx-on-all-bands)?

ChiefRA, 2025-03-29:
Thank you. I got it.