egzumer/uv-k5-firmware-custom #82: BusyCL - busy channel lockout. Busy is active as long as PTT is pressed.

ErikS-web, 2023-11-05:
Menu item 41. acts on internal PTT.
Description is _BusyCL - busy channel lockout, blocks radio from transmitting while signal is being received_
It show BUSY on screen when signal is received and PTT is pressed,

But transmission of the radio does "not start" when "received signal  stops transmitting" and BUSY stays on screen as long as PTT is pressed.

Have to release PTT and press it again to start transmission.

Would like to act like it is when external PTT (on external microphone) is used.

egzumer, 2023-11-05:
What would be the benefit of that? Either way you have to wait for the other side to finish. 

The external PTT works like that because of the hardware issue that causes the "busy lockout" like behavior. When press and hold the external PTT nothing is (or can be) detected in software. When the speaker is deactivated only then the pressed PTT is detected. It acts like you just pressed it. But again this is just because of the hardware issue.

ErikS-web, 2023-11-05:
The word "while" in the explanation suggests a different effect.

ErikS-web, 2023-11-05:
But... Why should you ever turn this option ON anyway 😇

egzumer, 2023-11-05:
You can turn it on, it works, but you have to press the button again when the signal disappears.

ErikS-web, 2023-11-05:
Either way you have to wait for the other side to finish and presse PTT again. Close item

closed 2023-11-05 by ErikS-web