egzumer/uv-k5-firmware-custom #85: DTMF minor issue

fastjetstream, 2023-11-06:
Hello,
I'm running egzumer v0.16, I assume there's a very minor bug with DTMF that affects only digit 6, all other digits generate perfect DTMF tone.
The problem is rather strange because I can hear the tone when I depress PTT + 6, but the tone received by the remote receiver is invalid... once again, all other digits generate valid tones, only 6 is incorrect... 
Any idea?
Thanks and regards
PS: congrats for your awesome piece of firmware, it's almost perfect and extremely appreciated :)

egzumer, 2023-11-07:
I imported some changes to DTMF tones from 1o11, there may be some error. How do you determine which one is good and which is not?

Robby69400, 2023-11-07:
> I imported some changes to DTMF tones from 1o11, there may be some error. How do you determine which one is good and which is not?

You can use a DTMF decoding app such as [https://play.google.com/store/apps/details?id=com.arcoirislabs.dtmfencdec.app&pcampaignid=web_share](url)

fastjetstream, 2023-11-07:
Hello, Thank you so much for your prompt reply.I’m working a repeater that uses DTMF to connect to other repeater networks, it’s a 2 digit code. I could switch to all networks that don’t have digit 6 in their dtmf code… if I send for instance code 96, I can hear the tones on my UV-K5, but I get a vocal message back “code 9 invalid” !! The server that handles the tones didn’t get the dtmf code for 6… (only 9 was ok, but since it expects a 2 digit code, I get the message code 9 invalid, so I know it recognized 9)All other digits work fine, if I send 95 or codes without 6, no problem.Best regards and 73’sON4SC

egzumer, 2023-11-07:
Take a look on github how your response look like, fix it.

fastjetstream, 2023-11-07:
I'm not sure the app will help, my understanding of DTMF is that when you press PTT + digit, the transceiver (uv-k5) sends the DTMF tone over the air (FM) and also sends a side tone to the uv-k5 speaker.
The side tone on the uv-k5 speaker seems fine, but for some reason, the dtmf tone that is sent on the air is invalid... I tried with a retevis handheld and no problem, so I assume the DTMF tone for digit 6 is not sent properly over the air.

egzumer, 2023-11-07:
Didn't you confuse codes? There was an error for 1, 3, A (679Hz instead of 697Hz). But others seem ok.
[image]

[dtmf_fix.packed.bin.zip](https://github.com/egzumer/uv-k5-firmware-custom/files/13282901/dtmf_fix.packed.bin.zip)

fastjetstream, 2023-11-07:
Thanks for your help, to give you some more details, I've now tested on a RRF hotspot, so I connect to the French RRF network ( http://rrf4.f5nlg.ovh:82/ ) where you can switch rooms with DTMF tones.
I connect to the RRF and can switch to any room using their DTMF code except one:
Here's the list:
 49 Reseau interconnecté du 49
 95 Parrot
 96 Call room
 97 FON (French Open Network)
 98 Tech chat
 99 International chat
100 General chat
101 Local chat
102 Experimental
103 Echolink
All DTMF codes are OK except 96 where I get a vocal message back code 9 not recognized because it didn't get the 6 right...
If you look at all the dtmf codes, all digits are used (0 thru 9), they all work except 6...
As stated, I can well hear the sidetone when I press 6 on the UV-K5 speaker, and I also monitor the frequency with my Retevis where I can also hear the 6 sidetone, but if I dial DTMF 96 on my Retevis handheld, it works just fine... I assume the DTMF for 6 is off, maybe slightly off, but enough so that it never works... when I connect to the RI49 network, I'm on another server and it doesn't work either, so it's not on just one installation, but everywhere. 
Is there anything in the code for this digit that would explain the problem?
Thank you so much for your help and the time dedicated to investigate, really appreciated :)
Best 73's, ON4SC

egzumer, 2023-11-07:
In the end it is all about audio frequencies. DTMF frequencies are as follows.
[image]
Form the audio spectrum that I attached earlier you can see that it all matches up perfectly. There is nothing more to be fixed besides 1,3,A that I already fixed.
You can use Audacity to record the audio of what your radio transmits and see what's there. Maybe there is some issue with your radio.

[image]
[image]

egzumer, 2023-11-07:
One more thing, what are the settings of the channel that you transmit with (frequency, ctcss, and so on)

fastjetstream, 2023-11-07:
Here's the settings:
Frequency: 145.2875 MHz simplex, FM Narrow, CTCSS = 74.4 Hz
I measured the audible sidetone for 6 and the 1477 Hz is spot on, I could not measure the 770 Hz because my equipment didn't go that low, so I assume the problem to lie with the 770 Hz
I can of course record an audio file of my 6 sidetone and send it if you want.
Thanks a million for your help and dedication :)

fastjetstream, 2023-11-07:
Just tested audio from speaker, got 771 Hz and 1477 Hz, so that looks spot on for digit 6 dtmf, could it be a level problem, I don't know and am really puzzled why it doesn't work.
Here's my dtmf settings:
[image]

egzumer, 2023-11-07:
The settings doesn't matter if DTMF frequencies are right (also there is nothing to change there). You can try to record both QS and the other radio you have and send me the audio file.

fastjetstream, 2023-11-07:
I will first try to re-install the original Quansheng firmware and see if it works or not.
If it doesn't work with the original firmware, I will close the case as the problem has not been introduced by the new code and I assume it's a Quansheng problem.
If it works with the original firmware, I will then record tone on QS speaker and received tone on my other HT, I will also record tone of Retevis handheld that works just fine.
Thanks a lot, you're more than helpful :)

egzumer, 2023-11-07:
Send the recordings either way.

Honusnap, 2023-11-07:
> t try to re-install the original Quansheng firmware and see if it works or not.
> If it doesn't work with the original firmware, I will close the case as the problem has not been introduced by the new [...]

Maybe the server is the problem ...

fastjetstream, 2023-11-08:
I've been redoing all tests very carefully on the UV-K5, I have DTMF problems for digits 2, 3 and 6, all other digits work fine... I first thought it was only digit 6, but 2 and 3 are also a problem.

To concentrate on digit 6, I started some tests on the RRF network with DTMF code 96,
I'm transmitting this code with the UV-K5 and get an error message back (looks like it doesn't get the 6) and with my Retevis and there it works fine.
Attached, the audio file of the UV-K5 where I record the speaker output while transmitting DTMF code 96 (it does not work) and the audio file of the same thing with my Retevis (it works just fine)
Please listen to all the way to the end.
When I have more time, I will record the UV-K5 transmission received on the Retevis and vice versa so it is possible to compare and will test with OEM firmware.

https://github.com/egzumer/uv-k5-firmware-custom/assets/148855760/38fa1c0f-7e93-4f07-a355-b8ebc75e34c0

https://github.com/egzumer/uv-k5-firmware-custom/assets/148855760/0186b2f9-38c7-402f-8871-9ba4efb86945

fastjetstream, 2023-11-08:
Here is the audio file of DTMF code 96 transmitted by uv-k5 on 145.2875 MHz as received by my Retevis (the recording is the audio output of the Retevis receive)

https://github.com/egzumer/uv-k5-firmware-custom/assets/148855760/0be09d0f-5b85-45b5-a485-89d372f99252

The RRF server sends an audio error message in French: "9 unknown command" because it fails to copy the dtmf of digit 6

egzumer, 2023-11-08:
It all looks good. The speaker sound is not important. What is important is what is being transmitted. The best test would be:
- use a 3rd radio as a receiver, place an audio recorder near it,
- don't move the recorder through tests so we can see if a volume of an audio changes,
- turn the transmitting radios volume knobs all the way down, not to record the transmitting radio speaker audio
- transmit 9,6,2,3 with retevis
- immediately after that transmit 9,6,2,3 with k5

I may not have time to work on this for the next 2 weeks so no rush, but I will come back to this at some point.
Maybe the modulation volume is too low, and rising the audio gain for DTMF transmission would help. We could try that out if I can find out how to do this. We'll see.

fastjetstream, 2023-11-08:
Thanks for your quick reply, for the last audio file, I typed DTMF 96 on the UV-K5 with volume all the way down and the microphone of the recorder was next to the retevis, so you really have what was transmitted by the UV-K5 but received on the Retevis.
I will make the requested tests with both k5 and retevis and send the audio files when I have time, as you say, no rush ;)
Thanks a lot and 73's

fastjetstream, 2023-11-10:
Hello,
I managed to get a 3rd radio to do the test :)
I've setup one of the Retevis on a fixed volume where I record the audio that is received from the other radios: another Retevis and the UV-K5 (all TX done on 145.2875 MHz simplex FM narrow)
I've transmitted DTMF 2, 3, 6 and 9 on both the Retevis and UV-K5 (volume set to minimum to make sure it won't be recorded) so the audio received on the other Retevis can be compared (only 2, 3 and 6 do not work on the UV-K5)
The audio file is attached :)
Best 73's
[image]

https://github.com/egzumer/uv-k5-firmware-custom/assets/148855760/fc75442a-024b-4720-8954-18229d38cc76

egzumer, 2023-11-10:
Great, I'm out for a week, I'll take a look at that when I get back.

fastjetstream, 2023-11-11:
OK, thanks, let me know if you need anything else, have a great time off :)

fastjetstream, 2023-11-17:
FYI, I've tested with a UV-5R plus and exactly same problem, so it's not something related specifically to my UV-5R...
[image]

egzumer, 2023-11-20:
Try with this firmware:
[dtmf_fix.packed.bin.zip](https://github.com/egzumer/uv-k5-firmware-custom/files/13417821/dtmf_fix.packed.bin.zip)

fastjetstream, 2023-11-20:
OK, will test tomorrow and let you know, thank you so much for your hard work and help

egzumer, 2023-11-20:
The issue was a difference in a volume between tones of the DTMF code. The repeater probably takes the max dB value of the signal and searches for tones that reach similar value, but the difference between tones in some codes was well over 5dB so it probably recognized those as single tone codes.

commit e0d80369 (2023-11-07): FIX #85: DTMF bug in 3 tones frequencies
commit 6c62490c (2023-11-20): FIX #85: Equalize tones volume in DTMF codes, fix wrong DTMF frequency values
commit 6e5e7e7a (2023-11-20): FIX #85: Equalize tones volume in DTMF codes, fix wrong DTMF frequency values
closed 2023-11-20 by egzumer