egzumer/uv-k5-firmware-custom #89: Battery Percentage doesn't correspond with Voltage

JMSCompSci, 2023-11-07:
I just installed the v0.16.  My battery voltage is at 8.05V.  The percentage reads 85%.  I have calibrated my battery to 1995 so it matches my multi-meter and I am using the 1600mAh setting.

ErikS-web, 2023-11-07:
First load your battery in docking station (for 24 hours or so)
The Li-ion battery loads until 8.4 volt. Than callibrate with 68.`BatCal` to 8.4 volt.
You can check the 8.4 volt at de to circle connections on the backside of the batterypack.

JMSCompSci, 2023-11-07:
Yes.  I did this with the older firmware.  That's how I arrived at 1995 being my calibration.  When I set it to 2200mAh setting, it displays 91% charge.

JMSCompSci, 2023-11-07:
I thought 8.05V is 100% charged isn't it?  It just doesn't display that.

egzumer, 2023-11-07:
> I thought 8.05V is 100% charged isn't it?

No it isn't

JMSCompSci, 2023-11-08:
What voltage do you guys get?

egzumer, 2023-11-08:
https://www.powerstream.com/lithium-ion-charge-voltage.htm

closed 2023-11-08 by egzumer