egzumer/uv-k5-firmware-custom #90: Need an adjustable "vox delay"

RoadRangerX, 2023-11-08:
Could you add an adjustment of the "vox delay" (the time it continues to transmit after you stop talking) to the menus? 🙂
I don't know what it presently is but the BlowFunguses generally have a long fixed delay of 1.5-2 seconds. To do packet radio on the cheap it needs to be more like 0.25 seconds using this cable:
https://www.amazon.com/dp/B01LMIBAZW](https://www.amazon.com/dp/B01LMIBAZW

RoadRangerX, 2023-11-08:
50ms or less would be ideal and get you worshipped like a god by the packet community :)

ErikS-web, 2023-11-09:
My packetstation uses a wired PTT. 
Interesting option to use the VOX on this radio for Packet-Radio and perhabs APRS.

The suggested cable costs more than the radio ;-(
Can be ordered cheaper or home-made.

ErikS-web, 2023-11-09:
Also a nice schematic at https://github.com/johnboiles/BaofengUV5R-TRRS

RoadRangerX, 2023-11-10:
Yes, that's basically what the cable I bought is. They work fine for APRS because APRS doesn't require listening for an ACK packet that could be transmitted before your radio has stopped transmitting if your vox delay is too long.

RoadRangerX, 2023-11-12:
Just wanted to add that some other radios let you set the delay to as little as 500ms - but this is reported to still be too long.

ErikS-web, 2023-12-15:
Interesting article https://blog.aprs.fi/2022/02/baofeng-btech-aprs-k1-iphone-problems.html

using the same PCB as picture above.

RoadRangerX, 2023-12-16:
Thanks for the link! I've switched to a PTT solution for now but this would still be awesome if we could get it to work well.

RoadRangerX, 2023-12-30:
On "0" I'm seeing about 1 second delay/tail, on "1" slightly longer?