egzumer/uv-k5-firmware-custom #92: almost perfect!

iz4hca, 2023-11-08:
only smeter to much open in AM (agc?), then would be perfect!
is the best firmware until right now!👍

rai4424, 2023-11-08:
I m very surprised that FW is improving day by day، with limited memory its a big acheavment to add awsome funtions

iz4hca, 2023-11-08:
> I m very surprised that FW is improving day by day، with limited memory its a big acheavment to add awsome funtions

absolutely!!👍

egzumer, 2023-11-08:
Thank you.
This is basically a duplicate of #53. Please go there, you will find a firmware there that displays raw RSSI value. If you have any way of telling what is the correlation between RSSI and correct value of dBm across spectrum (ideally 3 points per band) I could try to calibrate it. I'm not a ham operator and my knowledge is limited in this regard, also I don't know what kind of accuracy is satisfactory.
The dBm currently is calculated as `(rssi / 2) - 160` which comes from RF IC reference manual, but it looks like to me that it is shifted, and shows to high values.

iz4hca, 2023-11-08:
> Thank you. This is basically a duplicate of #53. Please go there, you will find a firmware there that displays raw RSSI value. If you have any way of telling what is the correlation between RSSI and [...]

  is enough to reduce a bit, not important be very precis.... try and we check...😉

egzumer, 2023-11-08:
Reduction in dBm
-30, // band 1
-36, // band 2
-54, // band 3
-48, // band 4
-48, // band 5
-48, // band 6
-12  // band 7

[reduced_rssi.packed.bin.zip](https://github.com/egzumer/uv-k5-firmware-custom/files/13301320/reduced_rssi.packed.bin.zip)

iz4hca, 2023-11-08:
> Reduction in dBm -30, // band 1 -36, // band 2 -54, // band 3 -48, // band 4 -48, // band 5 -48, // band 6 -12 // band 7
> 
> [reduced_rssi.packed.bin.zip]

so is too much, the pic show the difference,  in the middle of the last 2 release would be good👍

https://drive.google.com/file/d/1GmxOpAJev4ZpZjo76cbZLNbAKTJ6UXRL/view?usp=drivesdk

TNX MATE !

iz4hca, 2023-11-10:
Egzumer, the first reduce release is much hard, try this when you have a bit of time please🙏:
b1: -15 
b2: -15
b3: -25
b4: -20
b5: -20
b6: -20
b7: -7

iz4hca, 2023-11-20:
.. that's the right value! much more real, good job! 👍🙏

iz4hca, 2023-11-20:
...also the AM audio quality seem more clear, less noise 👌

egzumer, 2023-11-20:
> ...also the AM audio quality seem more clear, less noise 👌

Placebo effect, there is no change regarding that. Also the s-meter values are as you suggested, so good job for both of us.
Feedback from other users is also welcome.

iz4hca, 2023-11-20:
i said "SEEM" but only few minuts of testing, maybe placebo.... or maybe propagation🤷, anyway  the signal strength now is ok 👏👏

commit d37c3e35 (2023-11-20): FIX #92: Adjust s-meter values
closed 2023-11-22 by egzumer