egzumer/uv-k5-firmware-custom #94: Suggestion: Disable Tx with keyboard lock

SKUSKAskuska, 2023-11-08:
Hello - first to begin with - thanks for the great work! I would have one suggestion, I don't know if it is feasible.

It would be great if locking the keyboard (long press F) would also disable Tx via PTT.

Sometimes I carry the radio hanging on beltclip or in backpack only for Rx, and it is necessary to be very careful not to press the PTT while walking, because by default it is active even when the keyboard is locked.
Maybe some people prefer transmitting with keyboard locked, alternatively, on several of TYT handhelds it is solved by menu item for LOCK regime (key, PTT, key+PTT).

ErikS-web, 2023-11-09:
For now, you can use 61 `F Lock` - sets the TX frequency band plan (see [Hidden menu](https://github.com/egzumer/uv-k5-firmware-custom/wiki/Menu#hidden-menu))
and set it to "DISABLE ALL - disables TX on all frequencies"

karacsonyisandor, 2023-11-09:
Yes, but you have to enter the hidden menu every time to enable/disable... not practical.

ErikS-web, 2023-11-09:
As long as the radio stays ON, the hidden menu items can be accessed

karacsonyisandor, 2023-11-09:
@SKUSKAskuska  https://github.com/karacsonyisandor/uvk5-mods/commits/main/firmware.packed.bin 
moded the firmware so the hidden menu is always on. You can disable the tx with F Lock quickly.

SKUSKAskuska, 2023-11-09:
Great, thanks!

RoadRangerX, 2023-11-12:
Just wanted to say that I also would use this enhancement if it is ever implemented :-)

var-tz, 2023-11-15:
uvk5's ptt button is very easy to mistouch. sometimes i want lock it , sometimes i want speak someting,  use (lock option with tx) as a quick mode is a good idea.  use hidden menu is a bad idea.

gg33fr, 2023-12-31:
Hello, I don't understand the procedure to unblock in tx
I'm in the hidden menu, I go to F-lock, unlock all, I press M to validate and I have read manual, I come out and I go back to F-lock and unlock all or I do the procedure 10 times but it doesn't work, I always have tx-disable
Can someone give me the exact procedure with the keys to press in the exact order?
THANKS
Happy New Year

KvallVall, 2024-02-04:
A dumb workaround I use is to set the cursor on something in AM mode (can't transmit on AM) and lock the keypad.

BruMcD, 2024-02-12:
Mostly the keyboard lock is used to put down /pass the set out for use without you/users knocking themselves off channel etc in a car, bag or pocket and losing contact  (and it persists after switch off  /on for obvious reasons) 

The PTT / monitor /volume etc.   all still working on the right channel if needed is the whole point of it.

If locking everything is needed it should be another sort of lock to add to  the current on/off Lock