egzumer/uv-k5-firmware-custom #96: Request - siren alarm on FN2 after SOS

jepotter224, 2023-11-08:
For safety purposes requesting return of siren alarm on FN2 button press after the SOS.
Thank you for all the great features!

egzumer, 2023-11-08:
I don't quite get what you mean. Try google translate. If you wan't an alarm function you can add this in build options. QS already had it. Not an SOS, but still good for emergencies.

jepotter224, 2023-11-09:
Thank you for the quick review of this......the original fw when you did a long press of FN2 it would cause a siren noise to be emitted. Could this be available in the new FW please with the same long press on FN2?

egzumer, 2023-11-09:
There is a build option in the Makefile in this project that you can use to turn this option on.  I'm not going to add that, it is useless to me and it takes memory space. Read the readme, and the wiki, ask on telefram or facebook, community will help you. I don't have a time to be a personal support.

closed 2023-11-09 by egzumer

jepotter224, 2023-11-09:
Thank you for the guidance.