egzumer/uv-k5-firmware-custom #103: feasibility question about spectrum channel quantity

Robby69400, 2023-11-10:
Hello,
I was wondering it is possible to increase the quantity of channels selected in the spectrum.
My understanding is that you limit to 128x because there are 128 pixels so I would suggest to display only the strongest level in several channels when higher than 128x.
For example if I choose 1280x then I display the strongest signal in 10 consecutive frequencies. 
Of course performance will be affected but I hope it would still be enough to  monitor wider bandwidth. 
The max to this parameter the better as long as the radio continues to work properly.

I think such an evolution may interest a lot of listeners.
Thanks for reading me.

ErikS-web, 2023-11-10:
It is possible that the `SDR radio` project better suits your needs.

egzumer, 2023-12-19:
Partially implemented in 21 release

Robby69400, 2023-12-20:
Wow thanks so much, you're my Santa this year!

Robby69400, 2023-12-22:
So I have tested this feature, it's pretty good. Thanks again.
I have questions: do you know why it's faster than classical scan? 
I measured about 12ms per step in spectrum compared to around 100ms in scan.
Any possibility to accelerate more, for eg removing the spectrum ?

Do you think it could be possible to use up/down buttton to go to next "page" like in the original spectrum? For eg if I select 100M start to 110M stop, when I press Up it goes to 110-120M.

Anyway I'm already very happy with this.

kamilsss655, 2023-12-22:
It probably is slower in regular scan because it needs to load channel names from EEPROM each time it changes memory channel, wheras in spectrum mode it doesn't do that.

egzumer, 2023-12-22:
> It probably is slower in regular scan because...

This is incorrect, @Robby69400 has already asked this question before and got his answer.