egzumer/uv-k5-firmware-custom #105: re-introduce Voice Welcome (original under Menu 21)

ErikS-web, 2023-11-10:
Thanks for creating this nice firmware.

Originally, there is a 'Welcome' voice message when you turn on the radio in English or Chinese.
These settings appear to have been removed in the custom firmware.

I propose to reintroduce the option so that YOU can say "Hello world" or "Enjoy this hobby" to all of us ;-)

Robby69400, 2023-11-10:
I wonder if there is enough space for this...

ErikS-web, 2023-11-10:
For example, a special 'Christmas release' could also be released.

DualTachyon, 2023-11-10:
Voice messages are hard-coded in a dedicated chip. You cannot add new voices or change existing ones.

ErikS-web, 2023-11-11:
What dedicated chip ?
and ...
Should it be possible to reprogram the chip with the Voice of EGZUMER ?

DualTachyon, 2023-11-11:
Read the second line of my previous message.

ErikS-web, 2023-11-11:
What dedicated chip ?

DualTachyon, 2023-11-11:
The chip that this code drives:

https://github.com/DualTachyon/uv-k5-firmware/blob/main/audio.c#L166

DualTachyon, 2023-11-11:
https://t.me/quansheng_uvk5_en_dev/22/35709

ErikS-web, 2023-11-12:
Non-recordable IC. That's too bad.

Then I have to wish for something different for Christmas.

Maybe someone will come up with a tip

RoadRangerX, 2023-11-12:
I was recently noticing that the radio lacks a power-on sound and decided I'd rather it the way it is :-) .
Also wanted to thank DualTachyon, Egzumer et all for being programming gods :-) . I pretty much only use this radio now and my two DMR radios running the OpenGD77 firmware. Stock Chinese firmware kinda sucks eggs, eh?

ErikS-web, 2023-11-17:
Example found.... Enjoy this FUN https://www.youtube.com/shorts/Iws_SxRVX6M?feature=share

Not clear who made this ...

DualTachyon, 2023-11-17:
That person left telegram recently and all he is doing is reducing the voice play time. He's not playing custom audio.

egzumer, 2023-11-20:
As stated by others voices are hardcoded, and can't be changed. I personally don't like playing voices at all and I see it as great feature to be removed to save memory space for other things.

closed 2023-11-20 by egzumer