egzumer/uv-k5-firmware-custom #107: Scan lists

TokeTokerson, 2023-11-10:
Hello, appreciate all the upgrades thus far. The improved mic tx bar is a nice quality of life touch. Was wondering if the scan list selections can be retained upon turning the radio on. I've noticed that it defaults to scan list I instead of II or all memory channels when turning the radio off them on. 
- Thanks!

ErikS-web, 2023-11-10:
When you EXIT from the scan, the scan list setting is remembered
At Power-On is indeed list-1

Tip. Long press `SCAN` while channel scan is in progress, toggles scan lists 1/2/ALL(presented as *)

ErikS-web, 2023-11-10:
With Chirp you can adjust the default scan list.

[image]

Can you test that and report back the result?

ErikS-web, 2023-11-10:
In Menu the parameter can be set with 19. `SList` - _selects which channel is used by memory channel scanner_

closed 2023-11-17 by TokeTokerson