egzumer/uv-k5-firmware-custom #108: Turn off blacklight (BLMIN) when LowBattery.

chouyu, 2023-11-11:
as title

ErikS-web, 2023-11-11:
If you want to save energy, it is better **to do so in advance.** During use, make the setting as critical as possible for your use.

Your proposal does not seem ideal to me, because it would make the screen difficult to read for the user.

Just try out how the radio behaves when the battery is almost empty. Mine is somewhere between 6.3 and 6 volts. The radio starts to behave strangely and suddenly the radio screen freezes and TX is no longer possible.

chouyu, 2023-11-11:
When the radio is idle, a low battery alarm may not necessarily catch the user's attention. When BLMIN > 0, it accelerates battery consumption. By turning off BLMIN at this time, it can prolong the battery life until it is depleted.

ErikS-web, 2023-11-11:
Well, the warning plus LED on happend at my radio at 7.16 volts
Then there is still a small 1 volt difference, but charging the battery now is better.

[image]

egzumer, 2023-11-20:
Did you measure how much power would be saved?