egzumer/uv-k5-firmware-custom #110: I want to enable the Tx compandor globally

RoadRangerX, 2023-11-12:
Right now you have to do it individually on each channel - on the radio, CHIRP can't do it. I do think that CHIRP will not overwrite your change so at least there's that - unless you move channels around in CHIRP and then your change will move to another frequency LOL.

ErikS-web, 2023-11-12:
Just did some testeing. When12.` Compnd` (compander ) activated on VFO the C is present on all frequenties within that Band.

RoadRangerX, 2023-11-12:
I use channel+ frequency mode (as do most of us?).
I tried enabling it in VFO mode but that doesn't carry over to channel mode :'-(
https://wikidiff.com/compandor/compander

egzumer, 2023-11-20:
There is no point. Compander should be used on both sides at the same time so the per channel setting is the way to go. If I recall correctly you tried to rise the modulation levels, well this is not the purpose of this function. Modulation level is what it is on this radio, and I'm not going to mess with it with other functions that are not made for this. Maybe there is a modulation gain setting that could be changed, I'll take a look at that at some point.

closed 2023-11-23 by egzumer