egzumer/uv-k5-firmware-custom #111: Selectable dual- and single-band operation

ErikS-web, 2023-11-12:
A proposal for extra menu-option operation:  Dual/Single.

Can perhabs be combined with "RX-only".

Sometimes it is better to only be active on 1 band. I now solve this by setting both VFOs to the same frequency.

DualTachyon, 2023-11-12:
That's what Dual Watch is for. Turn it off and it stays on the VFO that you are on.

ErikS-web, 2023-11-12:
Indeed. Not tested before on the radio. But have a DUAL button on my Kenwood radio. That's how I came up with the idea.

It's under Menu 59 `RxMode` - sets how how the upper and lower frequency is used.
First option should do the trick: MAIN ONLY - always transmits and listens on the main frequency
Swiching Main channel works witch`2 A/B` as usual.

But in that setting I would prefer to see only 1 VFO / Channel on screen.

**************************************************************

Just updated Wiki Menu 59 `RxMode`  - sets how how the upper and lower frequency is used

   * MAIN ONLY - always transmits and listens on the main frequency (NO extra characters on screen)
   * DUAL RX RESPOND - listens to both frequencies, if signal is received on the secondary frequency it locks to it for a couple of seconds so you can respond to the call (**`DWR`** on screen)
   * CROSS BAND - always transmits on the primary and listens on the secondary frequency (**`XB`** on screen)
   * MAIN TX DUAL RX - always transmits on the primary, listens to both (**`DW`** on screen)

ErikS-web, 2023-11-14:
Added extra text to Menu 59 `RxMode`

>    Explanation of how these modes function
>    * The main channel is marked `►` and with PTT active `►TX`
>    * A receiving channel is marked `RX` as soon as a signal is received. Other VFO/channel is blocked from RX.
>    * With DUAL RX RESPOND, the secondary VFO/channel is marked `>` as temporarily the main channel when something is received there. 
>    * If nothing is received after a timer of 4 seconds (`><` on screen), that status will expire.

ErikS-web, 2023-11-16:
Made new [Issue 124 ](https://github.com/egzumer/uv-k5-firmware-custom/issues/124)

closed 2023-11-16 by ErikS-web