egzumer/uv-k5-firmware-custom #112: Fagci updates

Robby69400, 2023-11-12:
Hello, 
could you consider updating your spectrum to the latest from Fagci: 

https://github.com/fagci/uv-k5-firmware-fagci-mod/releases/tag/v1.2.1

There are some interesting evolutions such as band name indication (HAM, CB, AIR,...) ,

TX feature in the screen with PGA LNAs ...

RoadRangerX, 2023-11-12:
Just a comment from the peanut gallery - while the spectrum thing is a mega cool toy I don't find it of any practical use and wouldn't be sad if it was dumped in favor of some more useful use of the firmware space should such a thing come along :-)

tonurics, 2023-11-18:
There is a bit of a learning curve with the spectrum scanner, but once you understand how it works: it becomes a killer feature. Allowing for near instant scanning over a wide range frequencies.

egzumer, 2023-11-20:
The spectrum won't be abandoned as I like having it on my radio even though I don't use it that often. But there is a size issue which I mentioned many times before, so the exact copy of the current fagci spectrum is out of the question.

smokoks, 2023-12-26:
Вот  указание названия группы (HAM, CB, AIR, ...) было бы вообще отлично!!!

closed 2023-12-27 by egzumer