egzumer/uv-k5-firmware-custom #113: AM issue in spectrum

Robby69400, 2023-11-12:
Hello,
when i'm listening an air frequency in AM, the sound is good, but if I switch to spectrum and select AM, 
sometimes its OK sometimes unreadable. 
I think it happens especiallly with strong signals.
AM fix is activated

ErikS-web, 2023-11-12:
Can you add a foto ?

chouyu, 2023-11-12:
It  seems like AM-fix is not activated in spectrum mode.

ErikS-web, 2023-11-14:
Problem is the same as initiated from Discussions.

Will close this Issue. Pse continue under #118

egzumer, 2023-11-20:
The AM fix indeed doesn't work in the spectrum analyzer, it never did. This is something that needs to be worked on.

DL4SDR, 2023-11-26:
Hi, hmmm....AM  in AIRBand     in 0.15   works  here. actually  listening with 2 Radio's....one in VFO Mode, one in Spectrum-Mode
The Audio-Quality in Spectrum Mode  is not so good as in VFO Mode but readable  to identify everthing

closed 2023-12-03 by egzumer

jh4vaj, 2023-12-03:
It appears that reducing the LNAs/LNA/PGA appropriately can enable normal AM demodulation, in spectrum analyzer mode. 
I found this information on Twitter (X) (in Japanese), and it worked well with v0.19.

https://twitter.com/jr3kqf/status/1730911248001413373