egzumer/uv-k5-firmware-custom #114: improvement-make monitor mode independent on PTT

pawol, 2023-11-12:
improvement-make monitor mode independent on PTT

ErikS-web, 2023-11-12:
What monitor mode ?

I can only think of.. but that functions reversed.....

- PTT enters detail monitor mode when Spectrum-analyzer is active.

pawol, 2023-11-12:
I am talking about normal mode.  For me, the  spectrum is a gadget only. I do not compile it.

ErikS-web, 2023-11-12:
Aha you mean when the Monitor function is set under Custom Button 1 or 2.
Well this functions for **VFO.**

But there is a difference
When monitor is set open on a **Memory channnel**. Than is stays open after TX.
This is NOT on original firmware.

If you prefer continuously Open Monitor on VFO
Set squelch to 0 with
Menu option 60 `Sql` - squelch sensitivity level (0=OFF/1 .. 9)

pawol, 2023-11-12:
The issuevis not to set sql=0. It works in  way I described on 1of11 builds but last time his build are extremally bugged and cannot repport issues :(

ErikS-web, 2023-11-12:
Than use VFO-mode for now.

pawol, 2023-11-12:
VFO the same, PTT cancels monitor mode

pawol, 2023-11-12:
BTW. Squelch=0 makes monitor mode. No difference but much quicker

ErikS-web, 2023-11-15:
My testing that it worked with VFO was not correct. 
Squelch (re)opend due to QRM/interference from PC.

pawol, 2023-11-15:
So we have a bug

ErikS-web, 2023-11-15:
Yes, thats why I made that a separate issue.

I think it would be better to interpret your request as a request for an extension to Monitor mode.
Mode 0 behaves as set Squelch = 0. 
Mode 1 the current behavior: Squelch open but back to old level when TX is pressed.

commit ffdc0e03 (2023-11-23): FIX #114, #120: Restart monitor mode on PTT release, long press EXIT exits monitor mode
closed 2023-11-23 by egzumer