egzumer/uv-k5-firmware-custom #116: Bandwidth WIDE / NARROW

UW5EOT, 2023-11-12:
I noticed an interesting dependence of the received signal level on the bandwidth. When I change the bandwidth from "WIDE" to "NARROW", then those who barely opened the noise gate at level 5 begin to steadily open the noise gate and figure it out. Is it possible to add the ability to set the bandwidth separately for “TX” and “RX” in the menu? A particularly interesting option would be “RX-NARROW” + “TX-WIDE”
Thank you very much in advance for your work!

pawol, 2023-11-12:
It is notmal phenomenon to get different sensivity depending in channel width. Total nonsense to have different deviation in Tx and notbproper channel eidth on Rx

rai4424, 2023-11-13:
There is an idea that if we program  freq in two seperate chenels, one with Rx nerrow and other Tx wide. Tx from one chennel and receive from other BFO.
Note. Tx chennel should be used +- fuction to avoid receiving on same channel.

pawol, 2023-11-13:
It is easier to shift squelch level eg. 3dB depending on channel width to equalize the fillings

egzumer, 2023-11-20:
I observed other phenomenon, when receiving wide signals on a narrow setting the incomming signal becomes badly chopped up, almost impossible to understand.
I believe Yaesu have these options split but i don't see a point why to make it this way. You should use same options on both sides.

egzumer, 2023-11-20:
> There is an idea that if we program freq in two seperate chenels, one with Rx nerrow and other Tx wide. Tx from one chennel and receive from other BFO. Note. Tx chennel should be used +- fuction to avoid receiving on same channel.

You can use crossband option for that

closed 2023-11-23 by egzumer