egzumer/uv-k5-firmware-custom #117: MDC 1200

Zahurzahur, 2023-11-13:
Can we add MDC 1200 in version 0.17? 
Tx(ing) like Motorola.
If there is space isue then can we compress  FW 0.17?  as like Bricky has compressed a FW into 43.6 . I want to do that, plz guide me that is possible or not.

Bricky says:
13.2, GNU Linux. I use GCC 13.2 anyway so it's no big deal for me.

41.9K, but the recent MDC1200 port increased it to 43.6K.

pawol, 2023-11-13:
useless feature.

ErikS-web, 2023-11-13:
Who is **Bricky** ? Please help us and point us with an URL ?

Always interesting to watch.
But I do wonder in advance to what extent the performance is affected by the compression.

ETO-bigbang, 2023-11-14:
MDC1200 is very useful for exchanging information with more HAMs, and MDC can be used as an optional feature. If necessary, you can compile it yourself

Zahurzahur, 2023-11-14:
> Who is **Bricky** ? Please help us and point us with an URL ?
> 
> Always interesting to watch. But I do wonder in advance to what extent the performance is affected by the compression.

You can find it in contributers in this page as Bricky149

rai4424, 2023-11-14:
> useless feature.

Useful for motorolla users. We can able to use Uvk5 in group that is using motorollas.

ErikS-web, 2023-11-14:
> > Who is **Bricky** ? Please help us and point us with an URL ?
> > Always interesting to watch. But I do wonder in advance to what extent the performance is affected by the compression.
> 
> You can find it in contributers in this page as Bricky149

Please help us and point us to the information you mean with a URL?

Would be nice if you explain the difference.

var-tz, 2023-11-15:
very useful for  HAMs, the MDC1200 sound  especially easy to tell from noise.

egzumer, 2023-11-20:
I don't plan to add MDC1200. This is way to much effort for way to little gain. This is a really niche feature. If you just want the sound turn on the MDC in the Roger menu.

ETO-bigbang, 2023-11-20:
> I don't plan to add MDC1200. This is way to much effort for way to little gain. This is a really niche feature. If you just want the sound turn on the MDC in the Roger menu.

The MDC sound in the Roger menu sounds very fake. MDC1200 can be used as an optional feature, and HAM can compile it!

egzumer, 2023-11-20:
So what it sounds fake? Why do you want it to sound legit?

closed 2023-11-20 by egzumer