egzumer/uv-k5-firmware-custom #118: AM modulation only poor in Spectrum analyzer mode

ErikS-web, 2023-11-14:
### Discussed in https://github.com/egzumer/uv-k5-firmware-custom/discussions/101

<div type='discussions-op-text'>

<sup>Originally posted by **jh4vaj** November  9, 2023</sup>
It seems that the Spectrum analyzer mode in firmware v0.16 and v0.17 does not allow for AM (airband) demodulation. 
There are no issues with AM demodulation when not in Spectrum analyzer mode.

In v0.15, AM can be demodulated even in Spectrum analyzer mode.

Is there any other setting required for Spectrum analyzer mode in v0.16 and later?</div>

ErikS-web, 2023-11-14:
See also Issue #113. from [Robby69400](https://github.com/Robby69400) commented [2 days ago](https://github.com/egzumer/uv-k5-firmware-custom/issues/113#issue-1989375019)

> Hello, when i'm listening an air frequency in AM, the sound is good, but if I switch to spectrum and select AM, sometimes its OK sometimes unreadable. I think it happens especiallly with strong signals. AM fix is activated

egzumer, 2023-11-20:
This is a mess, I'm not going to jump between connected threads, please don't create issues that only mentions other issues.

closed 2023-11-20 by egzumer