egzumer/uv-k5-firmware-custom #121: The 1750 Hz Tone don't open the Relais

DL6JOE, 2023-11-16:
The 1750 Hz tone has no effect to the Relais. When I whistle I can open. I have tested the tone, it send out, but it has no effect.

ErikS-web, 2023-11-16:
I measured the frequency of the tone. It is actually 1744 Hz (measured on 2 radios with an Spectrum-App on 2 devices). 
To me, that seems within the tolerances of a 1750-tone slot. The signal is also loud enough.

[image]

However, a (1750) tone is only transmitted properly using the `internal PTT button`.
If you use an `external PTT`, the tone is shortened to a fraction of a second. This also applies to the DTMF tone codes.
But I think this is a hardware - and not a firmware - issue, because it also plays with the original firmware.
Behavior is the same on a PCB with revision V1.4 and a V1.6

How does that repeater behave with an different radio and 1750 Hz?
You might also test with a Tone-Generator(-App)

DL6JOE, 2023-11-20:
Hi Erik,

before, i used the original Firmware, and it works great, and also with another Mod. My different Radios work also fine with the repeater.

egzumer, 2023-11-20:
The frequency that is being transmitted is within +-1Hz. I see no reason why it would not work.

If you want me to check it, set another radio on receive. Set the quansheng radio volume to minimum so a speaker itself doesn't emit the tone. Record the sound that the other radio receives. Repeat this with the other Mod that works. Send me the audio files (use zip if it doesn't let you to attach your files)

DL6JOE, 2023-11-21:
Hi Erik,

i have tested it with your Firmware and another one. I have recorded the files. 
[uvmodwhosmatt.wav.zip](https://github.com/egzumer/uv-k5-firmware-custom/files/13423962/uvmodwhosmatt.wav.zip)

The sound from your Firmware is quieter and the signal narrower as the sound from the other firmware. The settings are both the same (Wide FM)

egzumer, 2023-11-23:
See if that fixes the issue:
https://github.com/egzumer/uv-k5-firmware-custom/suites/18447599160/artifacts/1069627369

DL6JOE, 2023-11-24:
Thank you, now it works great.

closed 2023-11-25 by egzumer

tuvok81, 2023-12-06:
hmm @egzumer is the link down for the fix?
gives in the future a fix direct on the firmware?

Cololagazelle, 2024-09-23:
Hello, i'm on egzummer 22 on my quansheng UVK5
When i try to use the 1750 tone (i press PTT and F2 button), it's like i'm just transmitting, but nothing else happens. With the sotck firmware, i could send the 1750 tone and i could hear the sound of the tone. 
Can you guys help me fix this?

Thank you very much

commit 9a3249a6 (2023-11-25): FIX #121: Rise tones volume