egzumer/uv-k5-firmware-custom #129: The main interface is chaotic and automatically jumps to 1300MHZ for monitoring

wu58430, 2023-11-21:
I previously used other customized systems. Before shooting this video, I burned the official system for a full reset, and then burned your system for a full reset. This video describes the problem I encountered.
Strangely, then I couldn't see any options in the 13th menu item (Demodu)
Thank you very much for your work
https://github.com/egzumer/uv-k5-firmware-custom/assets/52628935/2d9d8608-b434-4c32-af01-9fb1467f1c1e

wu58430, 2023-11-21:
I am using version 0.17
I wonder if this is related to the gEeprom section

egzumer, 2023-11-21:
What do you mean by "burned the system for a full reset", burning firmware doesn't reset anything.

wu58430, 2023-11-21:
> What do you mean by "burned the system for a full reset", burning firmware doesn't reset anything.

What I mean is to reset the system in the 69th menu after burning it

ErikS-web, 2023-11-21:
Menu 69 has 2 options... Please be more specific.

[image]

wu58430, 2023-11-22:
> Menu 69 has 2 options... Please be more specific.
> 
> [image]

anyone

wu58430, 2023-11-22:
[

https://github.com/egzumer/uv-k5-firmware-custom/assets/52628935/3adb2841-ff60-4b68-a024-88bc15d9e05e

](url)

commit 0ce7d7bd (2023-11-22): FIX #129: Bug in radio initialization after reset.
closed 2023-11-22 by egzumer